prusa3d/PrusaSlicer #1088: Crash on loading stl in 1.41.0-alpha2 and 3...

3Delights, 2018-07-28:
### Version
1.41.0-alpha2
1.41.0-alpha3

### Operating system type + version
Windows 10 Professional 64bit version 1803 (OS Build 17134.191)

### Behavior
* Slic3rPS crashes while loading a particular STL file, previous editions of Slic3r did not.  The 3MF version of the STL (created by MS 3D builder) loads OK.
* Add the STL from the ZIP below.
* STL should load as usual
* SLic3r crashes and closes with no messages.

[config, STL, 3MF.zip](https://github.com/prusa3d/Slic3r/files/2238324/config.STL.3MF.zip)

####

kazibole, 2018-07-29:
Possibly same as #1074.

FidelCapo, 2018-07-30:
Thank you for the report. I think this problem is caused by the Arrangement function. If the auto-center parts is enabled in the Slic3rPE preferences, this error occurs immediately when the object is imported. Can you please try import this object when the automatic arrangement is disabled?

 May I ask what application generated attached 3MF file?

Thank you.

3Delights, 2018-07-30:
I can confirm your suspicions Fidel, disabling "Auto-center Parts" in the preferences does indeed solve the problem.

FidelCapo, 2018-07-30:
Ok, thank you.

bubnikv, 2018-08-06:
Fixed in Slic3r 1.41.0-beta

closed 2018-08-06 by bubnikv