prusa3d/PrusaSlicer #1090: Feature Request - Gcode Import

ShamusTheTinker, 2018-07-28:
### Version
1.40.0+win64

### Operating system type + version
Windows 10 64bit

_Is this a new feature request?_ Yes

I'd like to have the options to import gcodes into slic3r.  If I use someone else's gcode file to print or if I forgot to save the profile for a slice that I did it'd be nice to import the file back in and be able look at it or modify it.

AliB10, 2018-08-01:
yes to this!
Cura has a great option which is load settings from gcode

bubnikv, 2018-08-02:
We would love to have that, but it is more complex that it seems:

1) You have to have the "gcode preview only" UI state.
2) You have to parse a general G-code.
3) You have to estimate the extrusion width / height and extrusion type.

Contributions are welcome to 2) and 3), the UI changes will likely have to be done at the HQ.

PMARINA, 2019-06-18:
Should this issue be closed? File >> Import >> Import Config >> (Select GCode File)?

closed 2019-06-18 by ShamusTheTinker

brosskgm, 2020-06-14:
Works in ideaMaker also. PrusaSlicer 2.2.0+Linux-64 : Ubuntu 18.04 64bit.
Not a PrusaSlicer / Slic3r PE generated g-code.
Even when it is a PrusaSlicer gcode it only shows it as a Print Settings, but does not display the item in plater.

Was told this is a good alternative to ideaMaker.

This feature is very important because many of us do get gcode from others or would like to re-look at what we did ourselves when we don't save every file 3MF.

Thanks