prusa3d/PrusaSlicer #1091: OctoPrint MetadataAnalysis Plugin

Daelus0, 2018-07-29:
Feature Request.

I'm really Loving the Accurancy of the new Prints Print Time Estimates, and It's great that we're adding this data to the gcode to update the display.

I've recently come across this neat little plugin for Octoprint from [Sven Lohrmann](https://github.com/malnvenshorn), [OctoPrint-MetaDataPreprocessor](https://github.com/malnvenshorn/OctoPrint-MetadataPreprocessor) which uses a postprocessing script and parses the gcode using the Octoprint parser on a computer to insert MetaData at the beginning of the file. It then uses a plugin installed on the Octoprint server which interputs the MetadataAnalysis to inject the information from the header. It's so much faster on a raspberry pi.

Looking at the gcode produced by Slic3r it seems to have atleast some of the information that this plugin creates/uses added at the end of the file now. I would love to see this accurate information brought into Octoprint rather then leaving it to make up it's own ideas.

Here's an example of the header added by Sven's plugin.
```
;OCTOPRINT_METADATA
;dimensions:
;  depth: 159.925
;  height: 30.5
;  width: 185.422
;extrusion_length:
;- 22149.996809992273
;extrusion_volume:
;- 53.27698586734601
;printing_area:
;  maxX: 185.422
;  maxY: 156.925
;  maxZ: 30.5
;  minX: 0.0
;  minY: -3.0
;  minZ: 0.0
;total_time: 250.87681661605896
;OCTOPRINT_METADATA_END
```

This segment here from Slic3r's output.
```
; filament used = 22128.5mm (53.2cm3)
; filament used = 61.2
; filament cost = 2.8
; total filament cost = 2.8
; estimated printing time = 8h 39m 49s
```
Is just showing a lower level of accuracy (though well with in tolerance), showing time as a string rather then an float, and doesn't show any dimensions of the printing object/area. These are both from the same print and Octoprint's Estimate is showing a little over 4 hours but Slic3r PE is showing 8h 39m 49s. And it did actually take much closer to the Slic3r PE estimate.

Daelus0, 2018-07-30:
If we can edit or better add a new line as well to get the time in minutes instead of as a string.

This line here in the file you're working with.
```
_write_format(file, "; estimated printing time = %s\n", m_time_estimator.get_time_hms().c_str());
```
I'm not entirely sure what to change to as I don't know what that function returns other then the obvious part that c_str() is most likely converting it to a string from something.... so I don't know if it's a simple as just changing the member calls... 

Then the only other thing we're missing is the dimensions and we can teach octoprint to look for those lines as they appear in a standard slic3r gcode export, saving the analysis time on an raspberry pi, i'm sure the Pi Zeros will love thank us for it. :)

stelgenhof, 2018-07-31:
In that case it's better to get the time in seconds if you're looking for accuracy :). The GCodeTimeEstimator class provides a method for that: `get_time()`.

The challenge here is that perhaps not every slicer provides this kind of metadata, hence the aforementioned plugin can cover all situations. On the other hand, the estimated metadata by OctoPrint may not be the same as that of those established by each slicer.

bubnikv, 2018-08-02:
I quickly skimmed throuh this thread, and I am not quite sure I understand the request.

Do you want Slic3r to export the OCTOPRINT_METADATA block?
Is there any formal definition of the OCTOPRINT_METADATA block?
Who interprets the OCTOPRINT_METADATA block?

stelgenhof, 2018-08-02:
@bubnikv The request is to have OctoPrint parse the Slic3r metadata (at the end of the gcode file), so that OctoPrint can display an accurate time estimate. Rather than OctoPrint's own estimation.

For large gcode files, OctoPrint can take some time to calculate the estimates. Having already the estimates (and other metadata) from the gcode file, the meta data can be displayed much faster in such a case.

The request for Slic3rPE is to make some small adjustments in the meta data so that the plugin can better parse/read this meta data. :)

bubnikv, 2018-08-02:
If there is a specific code in OctoPrint to parse the Slic3r metadata, why
cannot OctoPrint parse the hh:mm:ss format?

stelgenhof, 2018-08-02:
@bubnikv I agree: parsing the hh:mm:ss format is easy :) You may need to ask @Daelus0 about that.

Daelus0, 2018-08-02:
So, there's not currently anything in Octoprint directly, to my knowledge. There is a plugin which I've linked above. If we can export the data in the octoprint metadata format then we can just use that plugin. Either that or we can write our own plugin for octoprint to parse the metadata that Slic3r is outputting now. 

I've got 3 reasons for this request.

Octoprint running on a Rapsberry Pi takes a really long time to analyse gcode, and it would be best if we can just feed octoprint with all of the needed information in the gcode file.

Slic3r is way better and giving accurate estimates of print time and material usage, etc. So I'd love to get that imported into Octoprint instead of Octoprint wasting time resources trying to analyse the gcode and provide incorrect information. This is the biggest reason.

I'm also using Filament Manager and Cost Estimator. Filament Manager has an option to make it prompt me everytime I hit print for the filament selection, which is great because I almost always forget to change the filament selected before I print. But this needs to analysis to complete so that it knows, how many tools are being used so you can select the filament.

I'm not really clear on what the best option is. I'm thinking it's probably best to make our own plugin since that plugin isn't even in the public repo and requires manual install, though that's likely due to the fact that it requires a post-processing script run after the slicer. I've also had the thought that I could probably just write another post-processing script to follow up his and correct the numbers based on what's in the file from Slic3r.

bubnikv, 2020-12-03:
> If we can export the data in the octoprint metadata format then we can just use that plugin. Either that or we can write our own plugin for octoprint to parse the metadata that Slic3r is outputting now.

We prefer that this feature will be handled on the OctoPrint side. PrusaSlicer produces all the information, one "just" needs to parse it on the OctoPrint side. We would rather invest our time and effort into improving the slicing algorithms.
Closing.

closed 2020-12-03 by bubnikv