prusa3d/PrusaSlicer #1096: Strange Linear Advance behavior on MMU cooling moves

mcm001, 2018-07-30:
### Version
_Version of Slic3r used goes here_
1.41.0 Alpha 2
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 16.04, Windows 10
### Behavior
* _Describe the problem_ Linear Advance and slic3rPE 1.41.0 Alpha 2 gcode produces strange undesired results such as filament moving in the wrong direction (extruder negative on positive moves, or positive on negative moves)
* _Steps needed to reproduce the problem_
Run toolchange script with Linear_advance enabled and set to k=35
Observe filament behavior
https://youtu.be/iF9Jf3hIhdA is an example. Observe starting around 40 seconds, when loading occurs
The gcode run is attached as a ZIP so I could upload to github
[new_1_2.Gcode.zip](https://github.com/prusa3d/Slic3r/files/2242069/new_1_2.Gcode.zip)
You will see a toolchange snippet generated using mostly stock mmu2 settings with T0 and the header not present for testing purposes. 
* _Expected Results_ The extruder will move forward on ram, backwards on the 'yank', forward and backwards on the cooling moves, backward to parking position, forward again for loading (at all times). As you can see starting at 0:40 (where the load and wipe starts) the extruder starts slowly moving backwards and/or jerking backwards. I don't think it's skipped steps because the movement was smooth. Running with linear advance disabled removes this behavior, leading me to think the algorithm for linear advance is confused by wipe moves.

I would propose adding to line 595 of /xs/src/libSlic3r/Gcode/WipeTowerPrusaMM.cpp the following:
.append("K0"); 
or some other code to disable Linear Advance.
Following the toolchange filament Gcode (including linear advance settings) are sent so they will be reset there.
If the movement behavior is intended please let me know, but I have observed similar behavior both with Klipper's Pressure Advance as well as Prusa's Marlin Branch of Linear Advance 1.0

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Attached above I think?

lukasmatena, 2018-07-30:
Can you please check whether unedited Slic3r-generated gcode suffers from this issue?

Linear advance is disabled by M900 K0 that is emitted just before retraction on the object preceding the toolchange. It appears as if your gcode was created by copy and paste of only the moves on the tower, but linear advance is already disabled at that point. But I may be wrong.

mcm001, 2018-07-30:
It looks like in my files there is a K0 preceding the retraction prior to the move to the wipe tower. You are correct that linear advance is expected to be disabled. As everything seems to be working as designed I'm going to close the issue. Thanks for clarifying.

closed 2018-07-30 by mcm001