prusa3d/PrusaSlicer #1097: sudden stop and fast long extrusion

macdarren, 2018-07-30:
### Version
1.40.1+

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Mac OS 10.13.6

### Behavior
On a number of occasions I have had a print job printing normally when
the extruder will suddenly pause and eject a large glob of filament at high speeds.
This leads to a an obstruction on subsequent layers causing shifting etc.
Extrusion speed is very high...fast enough to strip the filament in the Bondtech extruder.

Slicing the model and printing is all that is required...sometimes minor changes
in slicer setting will shift where the problems occurs....it does not seem to happen
with every slice or model...

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
I failed to save my settings I will post them and the model when I can confirm
another failure....I am reviewing G-code for a known failure to be sure this is where the problem lies.

bubnikv, 2018-08-02:
Please provide the model / config. Also what is your printer & firmware?

macdarren, 2018-08-02:
Printer is a home brewed delta setup currently running Repetier 1.0.2 Firmware, although I had the problem with earlier firmware releases as well...I updated from the .9.x versions recently.

I think I may have found something further in that if I specify all the speeds and extrusion widths explicitly then the problem seems to go away....if I can get a reliable repeatable case I will try to post it....for now I am just specifying everything as I read that is how you do it at Prusa.  Often I like to try the full auto or auto on some items as it just in theory makes things easier.

bubnikv, 2019-01-29:
No model / config received, closing.

closed 2019-01-29 by bubnikv