prusa3d/PrusaSlicer #1099: Disable gcode generation by object

WarrenSchultz, 2018-07-31:
### Version
1.41.0a3/win64

### Operating system type + version
Win10/64

### Behavior
Feature request:
In the Settings right-click menu for an object in 3D view, it would be extremely useful to have a simple checkbox to disable gcode generation for that particular object.

Use case:
Testing out print settings on a group of complex objects that you have manually organized on the plate, and only want to print (or re-print) one object.

bubnikv, 2018-09-01:
As of now you may just move the unwanted objects completely out of the print bed.

WarrenSchultz, 2018-09-03:
That is handy, but it would be helpful to be able to disable the object in-place also. Super-low priority with the out-of-bed functionality, but also useful if you have a large number of small objects.

Alternatively, if there became group selection tools and movement, that would also help more than just this request.

bubnikv, 2018-12-12:
Slic3r 1.42 alpha to be released soon supports group selection, but currently you have to add the objects to the selection one by one. We plan to add a group selection by rectangle in the future.

bubnikv, 2019-08-07:
Implemented in the upcoming PrusaSlicer 2.1.0-alpha1
Closing.

closed 2019-08-07 by bubnikv