prusa3d/PrusaSlicer #1100: Bottom Fill Pattern not used if raft layers are enabled

lgv2018, 2018-07-31:
### Version
1.39.2

### Operating system type + version
MAC OSX 10.8

### Behavior
Nozzle = 0.8, Layer height = 0.5mm
Fill Density = 25%
Fill pattern = rectilinear 
Bottom/Top fill pattern = Hilbert Curve
Raft = 2 layers
Expected: "Bottom Layer" (first layer after last raft layer) to be filled with Hilbert Cube Pattern.
Actual: Bottom Layer fill pattern is same as "Fill Pattern"
[image]
. 
**Note - bottom/top fill pattern works if raft layer = 0.**
BUG

[image]
[image]
[image]
[image]

rsheasby, 2019-10-01:
I just ran into this same issue. I struggle to understand how this is intended behavior though? The help text says "Fill pattern for bottom infill. Only affects bottom external visible layer". Surely the bottom external layer is the layer on top of the raft, as the raft shouldn't count as visible if it's to be removed?

rsheasby, 2019-10-05:
After some more experimentation, it seems like the bottom pattern *does* still work, but only when the "Contact Z Distance" is set to 0. When set to anything non-zero, it stops working.

bubnikv, 2021-02-24:
Fixed with aee136cedd890ecb5072d9c9215a824fa3f365b1

1s object layer above raft will newly use normal flow, not bridging flow, and the bottom layer pattern will be respected.

closed 2021-02-24 by bubnikv