prusa3d/PrusaSlicer #1102: Top layer of model get changed with change in first layer height.

Anduiril, 2018-07-31:
### Version
Version 1.40.1+win64

### Operating system type + version
Windows 10

### Behavior
The top layer of bearing5.stl (Gear bearing by Emmet Lalish on Thingiverse) gets changed to a "solid" layer when the first layer is set to .3mm (printing at .2mm).  The 3d view displays properly but the preview and layer views have the top layer changed.  The problem goes away if I change the first layer to .2mm. in Print Settings.  This problem doesn't happen in regular Slic3r Version 1.2.9 or other slicers.  I've printed the file a few times before (I don't remember which slicer I used before) so I didn't check the preview and found out when the print came out bad.

Here is the preview at the top layer with .2mm first layer.
[image]

Here is the preview at the top layer with .3mm first layer
[image]

_Is this a new feature request?_  NO

#### STL/Config (.ZIP) where problem occurs
[Top layer corruption.zip](https://github.com/prusa3d/Slic3r/files/2245263/Top.layer.corruption.zip)

***This is my first bug report so if there is something I can do or do differently to make it easier on whomever is looking at this please let me know.

bubnikv, 2018-08-02:
Would you please try to fix your object through a netfabb service whether it helps?

Your reported issue may be the same as https://github.com/prusa3d/Slic3r/issues/895

bubnikv, 2018-09-01:
This is really the same issue as 
#831
#895
#1119
#1102

bubnikv, 2018-09-26:
Fixed with 1.41.1-beta.

closed 2018-09-26 by bubnikv