prusa3d/PrusaSlicer #1104: New GCode in Ver, 1.41.xxx slows my printer way down

BirkBinnard, 2018-07-31:
### Version
_Version of Slic3r used goes here_
All of the 1.41.xxx versions released to date

### Operating system type + version
Win10/64

### Behavior
The 1.41.xxx versions inserts this GCode at the start of the file:

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1500 T1500 ; sets acceleration (S) and retract acceleration (T), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

This code is not generated by Ver. 1.40.1

I'm not a GCode expert, but something in the listed code slows my printer way down. When printing it moves at about 20% normal speed. This includes the initial move from Home down to the print bed.

Note that my printer is a delta style, so Home is about 300 mm above the print bed center.

I could just delete the listed GCode from the generated file, but doing this every time is something I'd like to avoid. Is there a parameter I've missed that disables this code, or perhaps some way to tweak it so it does not slow down my printer?

Stoebei, 2018-07-31:
Hey. You have to set the Feedrate equal for all 3 Axis of a Delta Printer.  You can find this new Options in the Printer Settings, Machine Limits.

BirkBinnard, 2018-07-31:
Very interesting - thanks.

I changed the Z values for Max Feedrate, Max Accel., and Jerk Limits to be the same as the X & Y values. I'll test this when my current print is done - sometime tomorrow.

closed 2018-07-31 by BirkBinnard

Stoebei, 2018-07-31:
I just looked again at your Gcode.  I dont know what Delta Printer you have but 9000 Accel and 10 Jerk maybe could be to much.  Here are the Settings i use for my Anycubic Kossel Linear. Greetings
[image]

BirkBinnard, 2018-07-31:
Thanks for the heads up. My printer is an ATOM2 which has pretty high quality steppers. I'm not sure exactly what those various parameters need to be, but I'll do a test tomorrow or the next day.