prusa3d/PrusaSlicer #1105: Skirt missing in inner profiles

matt-3, 2018-07-31:
When the object contains nested parts, only the outer part gets a skirt, which is not enough.
See screenshot below, even though the inner bits are not connected to the outer bit, they do not get their own skirt. Consequently they do not adhere properly and the print fails.

[image]

from www.thingiverse.com/thing:2823006

### Version
Slic3r Prusa Edition
Version: 1.40.1+win64
Build:   Slic3rPE-1.40.1+win64-full-201807051331
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17134.167]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

bubnikv, 2018-08-02:
Duplicate of https://github.com/prusa3d/Slic3r/issues/491
Closing.

matt-3, 2018-08-02:
Thank you for checking but this is not a duplicate of #491. #491 deals with having a skirt on the inner side of the outer ring (ring = profile of the pipe-like object being printed).
I would like to have skirts attached to each (nested) ring in the screenshot I gave, not just an inner skirt for the outter ring.
I hope this explanation helps.
Thanks.

brawaga, 2020-05-16:
Agree, this is not a duplicate of #491, and I'd like to see this implemented too. Without it almost any complex thing printing will fail. I tried Oxford Bebe bricks, for example.

rtyr, 2021-09-21:
I think it is actually duplicate of https://github.com/prusa3d/PrusaSlicer/issues/61.

matt-3, 2021-09-21:
Correct, same as #61. Closing it. Thanks.

closed 2021-09-21 by matt-3