prusa3d/PrusaSlicer #1106: [Enhancement] Bypass "Start Printing on Upload" window when clicking "Send to Printer" Button

GreatGrizzly, 2018-07-31:
When I click on Upload to printer, I generally don't care what the file name is and usually want the print to start immediately.

I suggest adding a new checkbox below the brim checkbox in the right panel that bypasses the popup by using the default name and starts the print on upload.

The checkbox could be called "Start print immediately with default name" or something similar.

dikodahan, 2018-08-01:
I agree. in 95% (being conservative) of cases you are sending in order to print.
Maybe just a parameter in preferences if not the suggested option above but at list something to make this default stick.
I personally had multiple cases of sending without printing by mistake just because I missed selecting this option.

bubnikv, 2018-09-01:
@vojtechkral may we make the checkbox sticky?

BerndJM, 2018-09-09:
I think a pre-checked checkbox would be a good solution.
Even if there is already a print in progress nothing bad happens, Octoprint silently ignores the "print immediately" command.

commit d02c936b (2018-09-10): Print host improvements: presistent path & start print, bugfixes
Fixes problem with the path being scrolled to left on Windows
Fixes #1004
Fixes #1106
Fixes #1173
commit 8396bd84 (2018-09-10): Print host improvements: presistent path & start print, bugfixes
Fixes problem with the path being scrolled to left on Windows
Fixes #1004
Fixes #1106
Fixes #1173
commit 95b4d58c (2018-09-10): Print host improvements: presistent path & start print, bugfixes
Fixes problem with the path being scrolled to left on Windows
Fixes #1004
Fixes #1106
Fixes #1173
commit 4fa87490 (2018-09-13): Print host improvements: presistent path & start print, bugfixes
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1173
closed 2019-02-19 by vojtechkral