prusa3d/PrusaSlicer #1107: Incorrect dual material temperature management with ooze prevention enabled

molntapp, 2018-08-01:
### Version
1.40.1+Win64

### Operating system type + version
Windows 10 Pro 1803

### Behavior
* _Describe the problem_
If ooze prevention with temperature variation > 0 is enabled, both first-layer nozzle temperatures incorrectly are lowered. First layer is then printed with temperature intended for "other layers". This seems to be an issue only in the first layer. Printing the other layers "re-synchronizes" the temperature management and temps alter as expected.

* _Steps needed to reproduce the problem_
Load dual-extrusion STL. Use extruder 1 (T0) for infill and 2 (T1) for perimeters. Load two filament profiles both with first-layer temperatures 210, and other layer temperatures 190. Enable ooze prevention. Set temperature variation to -5 degrees. Slice and export g-code.

* _Expected Results_
First layer temperature on active extruder should be 210 when printing starts. First layer temperature of inactive extruder should be 205 (or 185) as long as it is raised to 210 before printing in first layer or 190 in other layers). Other layers temperatures should be 190 of active and 185 of inactive extruder.

* _Actual Results_
Intial temperatures of both extruders are set to 205 before printer specific g-code with M104 (don't wait for temp), and then again to 205 after the printer specific g-code with M109 (wait for temp). Then T1 is selected as first active extruder (which is correct in my setup). But its temperature is now set to 190 with M109 and the inactive extruder is left as-is at 205. Printing begins when the active extrude has cooled down to 190 from 205, i.e. with a to cold active extruder and a to hot inactive in the first layer.

#### STL/Config (.ZIP) where problem occurs
[config_gcode_and_stl.zip](https://github.com/prusa3d/Slic3r/files/2247470/config_gcode_and_stl.zip)

wilecoyote2015, 2020-04-14:
According to my findings, this happens also if the extruders are set to different temperatures. It seems that the temperature variation is applied to both extruders regardless which one is used. This also holds if the second extruder is not used for the print at all.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena