prusa3d/PrusaSlicer #1108: [Enhancement] Select which system profile to inherit from.

GreatGrizzly, 2018-08-01:
Give the user an option to select which system profile to inherit settings from. Currently the only way to change this is to recreate a new profile from the parent profile, or modify the config file directly by setting the "inherits=" entry.

bubnikv, 2018-08-02:
I expected it will come when we designed the profile inheritance. We decided on purpose to keep the initial implementation simple, and to add the bare minimum on request. Namely, we are afraid that support a full inheritance for every profile will add a significant complexity to the code (UI, configuration layer) and it will make the UI difficult to understand.

What would be your minimum requirements?

What exactly is your use case for this feature?

GreatGrizzly, 2018-08-02:
Here was my secenario: 

I have custom profiles for printer, filament and print settings. I export each profile in its own file. When I update the slic3r program, I get the latest manufacture settings. I then import the custom profiles.

The imported profiles lose their inheritance and thus I cant keep untouched settings automatically updated to the latest manufacturer settings.

Other Scenarios manually selecting parents to inherit from would be useful:

- If a system profile is deleted, there is no way for a user profile to easily change the inheritance to a new profile.
- Users would be able to inherit their own settings, making changes easier.

bubnikv, 2018-08-02:
> I have custom profiles for printer, filament and print settings. I export each profile in its own file. When I update the slic3r program, I get the latest manufacture settings. I then import the [...]

I suppose in this scenario Slic3r shall just maintain the link. I have fixed the same issue for the configs stored into the AMF/3MF files. Maybe it works now for you with the Slic3r-1.41.0-alpha3? If that does not work, then we may need to fix it.

> If a system profile is deleted, there is no way for a user profile to easily change the inheritance to a new profile.

That is a valid point, I wonder how probable is that. We should certainly strive to not rename the system profiles then.

> Users would be able to inherit their own settings, making changes easier.

I think this is the prime reason to ask for the inheritance editing. Do you just want to create your own "system" settings, or do you plan something more complicated? How complex you think the inheritance tree may grow?

GreatGrizzly, 2018-08-02:
My settings are as follows:
.05mm
.1mm
.15mm
.2mm
.25mm

These all share a lot of the same variables, but I have to duplicate the settings amongst them.

I have the following filament settings

Hatchbox PLA
Hatchbox PETG
Seinsmart PETG
Seinsmart TPU

The PETGs share a lot of settings, so having a parent PETG config would be ideal.

I also have different "printers":

MK3 TPU
MK3 PETG
MK3 PLA

This are identical except for their retraction values.

I see that you guys already have the right idea in the settings files, but it would be nice to have something we can use in the UI to set things up accordingly.

GreatGrizzly, 2018-08-06:
Imported settings losing their inheritance is an issue in Slic3r-1.41.0-alpha3.

bubnikv, 2018-08-07:
> I have custom profiles for printer, filament and print settings. I export each profile in its own file.

What exactly do you mean with "export each profile in its own file" ? You export a config?

> When I update the slic3r program, I get the latest manufacture settings. I then import the custom profiles.

You mean you load the config files (one print / filament / printer profile per file) one by one?

Why are you doing it? The update process should automatically keep your profiles intact.

GreatGrizzly, 2018-08-14:
I have my own git repository where I save my settings. I then use these settings on other machines when slicing models.

Before this auto update stuff, I had all my own profiles with the MK2. It was as easy as exporting config bundles and importing them on other machines. Now that settings are automatically updated, it messes with some of my custom settings when I import, one of the changes is my profiles loses its inheritance. If this feature was implemented. I can just assign a new parent, even if the parent is still the same as before.

vojtechkral, 2018-08-15:
@GreatGrizzly inside the `.ini`, the `inherits` property contains the parent. But be warned that this isn't public interface.

How does auto-updating mess your settings? If it does, that would seem like a bug.

You can also turn auto-updating off.

GreatGrizzly, 2018-08-15:
@vojtechkral

>inside the .ini, the inherits property contains the parent. But be warned that this isn't public interface.

My request is to make this public interface.

Vedrillan, 2020-09-11:
I do not have the exact same issue and need but it probably shares a pretty close root cause, as it seems to be driven by the inheritance behaviors.

I wanted to do a bundle config ini file to manage my custom profiles with heavy use of the inheritance feature to make it as clean as possible but after breaking a few teeth on that I ended up here :V 

This is for instance what I wanted to create in a bundle config file:
```
[filament:PETG low temp]
inherits = Generic PETG
first_layer_temperature = 210
temperature = 220

[printer:Original Prusa i3 MK3S - octoprint]
inherits = Original Prusa i3 MK3S
host_type = octoprint
print_host = 192.168.1.42
printhost_apikey = 1234ABCD
```

I though this would create a new printer and filament profiles with all inherited values set dynamically from their parent profiles, but this is not the case, instead, all unspecified values are set to the default values. The behavior in `%APPDATA%/PrusaSlicer" files is similar. Basically the `inherits` field does not mean that all unspecified values will be inherited from the parent profile, it behaves more as a `copied_from` field (the GUI seems to correctly traverse the inheritance tree though with the lock indicators).

Right now with every system profile changes I have to search for what changed and apply the changes as well to my "inheriting" user profiles, which is rather tedious, so having a more dynamic inheritance system could be helpful.

murk-sy, 2024-07-02:
+1

This isn't commonly done, but transferring non-IS based profiles to IS based profiles would be much simpler if parent/inheritance can be changed.

Another situation where I would like to use it is for "transferring" specific settings between printers - i.e. moving a Mini print profile to the XL and vice versa.
It's much easier to transfer everything and then manually go through the menus to reset settings that aren't necessary for you to default. Profile comparison is okay for checking a few things but if you need to be through, doing it like that is much easier to work with.

I know it can be done manually by editing ini files, but I'd prefer if there was a button.

closed 2026-09-29 by prusaslicer-bot[bot]