prusa3d/PrusaSlicer #1110: [Enhancement]Use git repo for config directory

Tirpitz93, 2018-08-01:
### Version
Future

### Operating system type + version
all
### Behavior
It would be cool if you could set up a (private) git repo to use as a config directory which is automagically pushes when settings are updated and pulled when slic3r is started.

This is a feature request
I know this is another obscure request from myself :D

bubnikv, 2018-08-02:
I am sorry, but I am quite sure that we at Prusa3D will not implement such a feature. I am marking this request with a "volunteer needed" flag for reference.

supermerill, 2018-08-02:
Well, you may be able already do that yourself.

Create a script file which do 
* a git pull into your setting folder repo
* launch slic3r with perl slic3r.pl --datadir /path/to/my/setting/repo
* a git push when slic3r return

(untested) example:

```
cd /home/supermerill/slic3rSettings
git pull
perl /home/supermerill/slic3rPE/slic3r.pl --datadir /home/supermerill/slic3rSettings
git add *
git commit -a --allow-empty-message -m ''
git push
```

It should be better if you can create a small automatic message for the commit, with the date for example.

tracernz, 2018-08-02:
It would be nice if Slic3r had hooks so you could have your own scripts run when e.g. on launch, when settings are saved. Then you could implement this yourself fairly trivially with whatever VCS or backing store you desire.

closed 2026-09-29 by prusaslicer-bot[bot]