prusa3d/PrusaSlicer #1111: Purging into infill and Purging into objects has no effect

jimmy00784, 2018-08-01:
### Version
Slic3r PE 1.41.alpha 3

### Operating system type + version
Arch Linux 64 bit

### Behavior
Purging into infill and Purging into objects has no effect

Steps to recreate:
1. Add STL to the platter.
2. Go to STL settings.
3. Click Load Generic to create a box of arbitrary dimensions.
4. Position the generic object to overlap the original STL partially.
5. Change extruders on both STL and generic object to be different.
6. In object settings, check "Purging into infill" or "Purging into objects"
7. Click Ok
8. Switch to Preview to preview GCODE rendering.
9. Export the GCode and preview in other tools like Pronterface etc.

Observation:
Purging into infill and Purging into objects has not visible effects on the generated GCODE.

Settings view
[image]
Platter Preview
[image]

Please provide known settings and any testing related instructions if possible to eliminate user error.

lukasmatena, 2018-08-02:
Thanks for reporting, but I didn't manage to reproduce the issue. Setting 'Purging into infill' on the object correctly uses the infill for purging. It is even possible to only purge into infill of one of the parts only. Setting the whole object as sacrificial for purging results in no wipe tower and no toolchanges (as long as there are no other objects on the plate) - that is also correct, since Slic3r understands you don't care what the color will be, so it doesn't bother with toolchanges.

The problem is most likely that as of today, the preview shown after clicking the Slice Now button is not correct as for the colors. It is necessary to 'Export GCode' first to get the actual preview. I understand that this is an annoyance, but this is due to Slic3r internal architecture and can be only solved by quite substantial code-shuffling, which will take some time - we are planning to do it soon, though. It is the same reason why you only get the print time estimate only after gcode export. But I understand that this problem should be better documented.

Looking at your screenshot, I assume it is correct too - there is a sparse section on the wipe tower, that would normally not be there (as long as purging volumes for 1->2 and 2->1 toolchanges is the same). The object is small, so it does not save much. Also, only one half of it is used for purging, so the perimeter-infill order is not violated. Just Export GCode and we'll see. Using external gcode viewers is (at least from my experience) very unreliable.

Please, let us know if that was the problem.

jimmy00784, 2018-08-02:
Ok, so the wipe tower still gets created but with reduced filament 
wastage if I set Purging in infill?

bubnikv, 2018-08-02:
yes

jimmy00784, 2018-08-02:
Thanks for the clarification. I'll close the issue.

closed 2018-08-02 by jimmy00784