prusa3d/PrusaSlicer #1112: infill and first top layer height at 1mm even though I'm asking for 0.4mm

dushyantahuja, 2018-08-02:
### Version
1.40.1+

### Operating system type + version
Mac OS High Sierra
v 10.13.6

### Behavior

* I'm trying to slice an STL file with 0.4mm layer height, with gyroid infill. The outer perimeters show a height of 0.4mm; however, the infill and the first top layer has a height of 1mm. This causes the top layers to crash into the head and the print gets spoiled. 

[image]
[image]

I'm using an E3D volcano with a 1mm nozzle. Combine infill is set to 1 - so not sure what's happening here. 

The expectation is that the infill should have the same height as the rest of the model. 

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/2252516/Archive.zip)

supermerill, 2018-08-02:
Lucky one, the "height" option works for you.

Speculation: if it's the first solid layer for the "top layers", it's sliced as a bridge, hence the hight that is equal to your nozzle size.

As a temporary fix, you can reduce the bridge flow and see if it prints better.

dushyantahuja, 2018-08-02:
Is the infill and bridge height always equal to the nozzle width !!

supermerill, 2018-08-02:
The first solid infill over sparse infill is a bridge.

>Bridge height always equal to the nozzle width 
Of course, as there are no support to push against, your nozzle will extrude a round filament. You can reduce the bridge flow to hope it will create a nice ellipse. I imagine that with a 1mm nozzle, you have to test which bridge ratio work the best.

dushyantahuja, 2018-08-02:
Thanks - will try that. However, that would not explain why the infill has a height of 1mm. Tried to slice it with a 0.4mm nozzle - and have the same issue. The infill and first top layer are 0.4mm where the slicing profile is for 0.2mm.
[image]

dushyantahuja, 2018-08-02:
Looks like an issue with the gyroid fill - tried honeycomb fill and that came up with 0.4mm height for all the layers (except the bridge layer)
[image]

supermerill, 2018-08-02:
it's a problem with the implementation in the current branch of slicµ3r pe : the gyroid infill is sliced as a bridge because it's not completly over the previous layer. But i think personally it should be removed. 

Look at the 3Dhoneycomb.

bubnikv, 2018-10-23:
The bridging flow was disabled for Gyroid infill with b429b39
The fix will be in 1.41.1 final.
Closing.

closed 2018-10-23 by bubnikv