prusa3d/PrusaSlicer #1113: [REQUEST] Please add a GUI field to set wipe distance

aaronevers, 2018-08-02:
Please add a GUI field for initial wipe distance. I'd like to double it for PETG without having to edit G-code myself.

aaronevers, 2018-08-02:
Conditional G-code in custom start G-code is a workaround not a solution.

bubnikv, 2018-08-02:
You can do that with a conditional G-code in your custom start G-code.
something like this will work:
{if filament_type == "PET"}
your wipe line
{else}
another line
{endif}

So you insist, that we shall implement your request.
Let's wait for other's reaction. If there is none, I will close your request in about a month.

aaronevers, 2018-08-02:
It's a feature request, not a bug fix. Accept the suggestion or not, but closing with a workaround doesn't address my request.

bubnikv, 2018-08-02:
We have to face reality. You will not implement your request, and we don't have resources to handle all the 353 open issues. And if nobody is ever going to implement your request, then your request remains just a noise in the list. Also it is dangerous to clutter the application with gazzilions of configuration values and switches just because one person asks for it.

aaronevers, 2018-08-02:
This really isn't the discussion I meant to bring up. I'm sorry for cluttering your github, but I think it's a useful feature and you closed it without any discourse. Please don't pre-judge your users.

aaronevers, 2018-08-02:
No, you're right. If someone else has the same issue, they can open a new ticket. Please link it to this so I get a notification.

closed 2018-08-02 by aaronevers