prusa3d/PrusaSlicer #1116: Disappearing Slic3r Window after disconnecting from second monitor.

baokhangnn, 2018-08-02:
### Version
Slic3r Prusa Edition - 1.40.1+ (Mac)

### Operating system type + version
Mac OS 10.13.6 (High Sierra)

### Behavior
Slic3r window does not re-locate to main display after disconnecting from secondary monitor.
* _Steps needed to reproduce the problem_

1. Laptop connected to external monitor.
2. Launch Slic3r and place window on external monitor.
3. Quit Slicer.
4. Disconnect external monitor.
5. Launch slic3r.
6. Slic3r doesn't show up on main monitor. However when collecting windows it zooms in from where the external monitor was located.

_Is this a new feature request?_ No. 

This is probably related to Issue #129 and #925.

bubnikv, 2018-08-02:
@FidelCapo Would you please try to reproduce it?

baokhangnn, 2018-08-03:
Hello,

I can do this over and over all day long. I just have to remember, before I close Slic3r that I have to move it to my laptop screen. Then I can unplug the external monitor and still get to slic3r.

However, If I have Slic3r open and pull the external monitor then it relocates to my laptop monitor just fine.

Thanks,

-Bao

bubnikv, 2018-08-03:
We were able to reproduce it on both OSX and Windows. We hope to fix it for the final release.

baokhangnn, 2018-08-03:
Thank you!

You guys really are amazing!!!

-Bao

commit 2a851025 (2018-08-10): WIP Fix #1116
commit ce6cabeb (2018-09-04): WIP Fix #1116
commit c4ce4ad6 (2018-09-04): WIP Fix #1116
commit c9d40219 (2018-09-05): Fix window size persistence
Fixes #1116
closed 2018-09-14 by bubnikv