prusa3d/PrusaSlicer #1118: Raise z axis all the way when print is done for easy bed plate removal

ayberkozgur, 2018-08-02:
Very simple feature request: Raise z axis all the way up when print is done for easy bed plate removal. Right now, the plate bumps into the extruder especially for short prints.

dikodahan, 2018-08-02:
+1 for that!
Every time I have a low profile print I need to go through the menu to raise the extruder so I can easily remove the plate on my MK3. Very annoying.
Not sure about all the way up, but definitely at least to the middle of the Z axis, or 75% up.

p3ck, 2018-08-02:
Are you aware you can hold the dial button down and it will jump to the
Z-axis control?

ayberkozgur, 2018-08-02:
@p3ck Yes, but ideally this manual intervention should not be necessary at the end of every print.

GreatGrizzly, 2018-08-02:
I have custom gcode that squirts out some plastic then raises the print head to the top. This guarentees that plastic wont curl up and stick to the nozzle when it cools down.

I think both of those features should be standard.

bubnikv, 2018-08-03:
Our current end G-code looks something like this:

```
G4 ; wait
M221 S100
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up
G1 X0 Y200; home X axis
M84 ; disable motors
```

See the line
```
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head 
```
which is responsible for lifting the print head at the end of the print by 3 cm if there is space. I suppose this implementation fulfills your needs, right?

ayberkozgur, 2018-08-03:
@bubnikv Exactly, I have 1.39.0 from the Ubuntu repos so I did not see this change yet. However I would argue that z should be raised a clean 100mm for zero chance of hitting the extruder as the magnets are pretty strong and quite a lot of energy can be built while pulling the bed up.

bubnikv, 2018-08-04:
> However I would argue that z should be raised a clean 100mm for zero chance of hitting the extruder as the magnets are pretty strong and quite a lot of energy can be built while pulling the bed up.

I think your argument is best to be confirmed by experiment. Did you try our 30mm lift?

Anyway, this value is configurable in the printer profile. Are your needs satisfied? Can we close this issue?

GreatGrizzly, 2018-08-07:
See #1126. I have end code that handles this.

I believe that the printer should have max clearance after a print, not only to take the heat bed off, but to allow easier access to hotend to get plastic off that may have stuck to it during a print.

ayberkozgur, 2018-08-07:
@GreatGrizzly I totally agree about the clearance and the extra extruded plastic, it's quite a shame how badly the extrusion before the intro line is currently handled. It always results in extra bits of plastic that ruins the first 2-3 layers. Therefore I close this issue now and subscribe to #1126.

closed 2018-08-07 by ayberkozgur

ayberkozgur, 2018-08-14:
@bubnikv As a side note, I've been trying out building `master` directly from the repo and I can say that 30mm is not enough. With shallow prints, the bed easily hits the extruder due to the strength of the magnets.