prusa3d/PrusaSlicer #1119: Additional non-support geometry being added to model

kdorff, 2018-08-02:
### Version
1.40.1+

### Operating system type + version
Mac OSX High Sierra 10.13.6

### Behavior
Slicing the model https://www.thingiverse.com/thing:2454444 at 0.20mm, 20% infill, Slic3r inserts solid layers toward the constriction of the model. Support is set to None. No brim. These appear in layers at 48.80, 55.40, 58.40, 58.60, 61.80, 62.00, and possibly others (I got these numbers of scanning through the layers on the Layers view after slicing).

This geometry doesn't appear in the model itself. When I sliced with Cura, I don't see these. When I printed this before, I assumed it was some sort of support material, but it does NOT break out. So now I am thinking it's a bug.

### Screenshot
View from below with the base of the model clipped away so you can see the red geometry that is being added to the model from seemingly nowhere.

[image]

#### STL/Config (.ZIP) where problem occurs
[problem-stl-and-config.zip](https://github.com/prusa3d/Slic3r/files/2255463/problem-stl-and-config.zip)

kdorff, 2018-08-16:
This still happens with `1.41.0-beta+`.

bubnikv, 2018-09-01:
I bet it is the same issue as #1166, #1158 #831 #895 #1102
I am looking into a fix.

bubnikv, 2018-10-05:
It will be fixed in 1.41.1-rc. Closing.

closed 2018-10-05 by bubnikv

kdorff, 2018-10-10:
I've downloaded -rc and confirm the extra geometry is no longer present.

commit 517b96ee (2018-10-05): Yet another improvement in closing gaps in slices.
Fixes #1256 and it finally fixes #1119 as well.