prusa3d/PrusaSlicer #1121: Issue with quad extruder setup with single nozzle

jimmy00784, 2018-08-04:
### Version
1.41.0 alpha 3

### Operating system type + version
Arch Linux 64 bit

### Behavior
When attempting to use 4 filaments in a multi-material single nozzle setup, initial purge escapes outside of the print bed of dimensions 200mm x 200mm.
I couldn't find any setting that would allow configuring the position of dimensions of the purge.

[image]

lukasmatena, 2018-08-04:
@bubnikv: "The initial purge is as wide as the wipe tower."
This is no longer true. The width calculation was changed while fixing https://github.com/prusa3d/Slic3r/issues/571 and preparing for MMU2 that might want to prime up to five tools. It was still left Prusa-specific and assumes the bed width of 240mm. I will change it for the upcoming release so it uses the actual dimensions of the bed. Not sure what to do with circular beds, though - maybe prime along the diameter and let the user remove it before actual print.

This is probably not the last thing that is Prusa specific. We should decide on how to provide MM support for non-Prusa printers.

jimmy00784, 2018-08-04:
Could this be resolved by priming within the Wipe Tower itself?

jimmy00784, 2018-08-23:
@bubnikv Any idea when we can expect a fix for this issue?

bubnikv, 2019-08-01:
> Any idea when we can expect a fix for this issue?

What is your printer? Do you still have the issue with the PrusaSlicer 2.0.0?

jimmy00784, 2019-08-14:
The printer is customized Anet A8 with a multi material single nozzle setup.
The issue is still not fixed in PrusaSlicer 2.0.0
[image]

Project file attached:
[QuadExtCalibration.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3502391/QuadExtCalibration.3mf.zip)

lukasmatena, 2019-08-14:
It should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/1d20f40defc926301ba8aaf403cba44b02e515b7. The commit is now only in a feature branch lm_circular_priming. We'll see if we push it into 2.1.0 release or not.

The commit makes sure it chooses the priming section width based on actual bed width, not assuming anything. I also placed the priming lines along the diameter on circular beds, but it uses no clever logic. Sadly, we don't have time to support everything that exists (or might exist).
[image]
[image]

lukasmatena, 2019-08-20:
The fix was just merged into master branch to be a part of the next release (2.1.0). Please, retest after it is released (we should see a beta this week) and let us know if everything is ok.

I'm closing the issue.  We can reopen it if needed.

closed 2019-08-20 by lukasmatena