prusa3d/PrusaSlicer #1123: program crashing when saving a preset

scanlight, 2018-08-05:
### Version
1.40 but also earlier versions

running windows 10 64x

program crashing (no warning just disapears) when trying to save a preset.
started a few months ago. today i tried the newest version but the same problem

all works flaweless and can export gcode.
until i try to save a modified preset.
then its gone

FidelCapo, 2018-08-06:
Can you please upload config file with that modified preset (File - Export Config...)? Thank you.

scanlight, 2018-08-06:
Thats the problem i cant save any present.
It loads the default settings. When i modify anything an click the save present button the program crashes.
I dont know of there is any log file to get more info of whats happening at the moment of the crash

FidelCapo, 2018-08-06:
Does this save button mean the floppy disk icon when selecting presets?

bubnikv, 2018-08-07:
What exactly is happening? Does slic3r close? When exactly?
Is there anything special about your directory where Slic3r profiles reside? Isn't it a network drive? Did you set a reparse point somewhere on the path to this directory?

bubnikv, 2018-09-01:
Are you technical enough to install a Microsoft SDK with a debugger, or Visual Studio 2013, and provide us with a call stack?

scanlight, 2018-09-01:
Never done this but am gonna do my best.

Verzonden vanaf mijn Samsung Galaxy-smartphone.

bubnikv, 2018-09-01:
@vojtechkral would you please describe how to run Slic3r under a debugger?

here are the symbols for the upcoming 1.41.0 release.

[Slic3rPE-1.41.0+win64-full-201809010758.pdb.zip](https://github.com/prusa3d/Slic3r/files/2342229/Slic3rPE-1.41.0%2Bwin64-full-201809010758.pdb.zip)
[Uploading Slic3rPE-1.41.0+win32-full-201809010758.pdb.zip…]()

bubnikv, 2018-09-01:
[Slic3rPE-1.41.0+win32-full-201809010758.pdb.zip](https://github.com/prusa3d/Slic3r/files/2342231/Slic3rPE-1.41.0%2Bwin32-full-201809010758.pdb.zip)

vojtechkral, 2018-09-03:
@scanlight Hi, I've created a short tutorial on how to produce a crashdump: https://github.com/prusa3d/Slic3r/wiki/Enabling-crashdump-collection-on-Windows

Would you please be so kind as to create the crashdump and share it with us so that we can have a look at what's going on in there? Thanks!

scanlight, 2018-09-14:
finally found the reason of the error.
seems like when i save the preset it want to write to a file. but the file doesnt exist. thats why the software is crashing.
i exported the presets to a new file name now i can save the presetr as usual

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv