prusa3d/PrusaSlicer #1124: Feature request: Add Prusa Slic3r to Gnome Software

mrjohnc, 2018-08-05:
Adding Slic3r Prusa Edition to Gnome software would make it easier to install and keep up to date on Linux distributions using Gnome (which many of the most popular Linux distros use)

https://wiki.gnome.org/Apps/Software

bubnikv, 2018-08-05:
Slic3r is written using wxWidgets toolkit over GTK2 (or GTK3), so it is officially not a Gnome application. Also there is a Slic3r PE package for Debian, Ubuntu and likely other distros, so if you don't like the AppImage format we provide, you can likely install Slic3r PE from your distro, though it will likely be a bit older version.

Now I don't understand the request. Why should there by a Gnome app repository, if the distribution model for applications on Linux is the distro?

Maybe @vojtechkral has an idea?

mrjohnc, 2018-08-05:
Sorry I'm not explaining very well, basically I want to be able to install Prusa Slic3r from the sofware centre thing that comes in Ubuntu (like you can do with Cura). Whilst I know how to install the software in the other ways it would lower the barrier to others to be able to do this and would allow people to keep it up to date easily.
[image]

vojtechkral, 2018-08-06:
Ubuntu Software Center is discontinued. Gnome Software is active, but it is, as far as I know, more or less just a frontend to distro package managers. So basically better package presence would be likely needed.

grigorig, 2018-08-10:
I think what OP might really mean is that Slic3r PE should use a standard distribution mechanism, such as Snap or Flatpak that is portable and independent from a specific distribution. Sort of like AppImage, but more suitable for the "app store" use case and with better security/isolation.

mrjohnc, 2018-08-10:
@grigorig yes this is exactly what I meant, thanks very much

mrjohnc, 2018-08-11:
@bubnikv feel free to rename to how @grigorig explains

mrjohnc, 2018-08-18:
Coupled with this it would be really great to have the software know when it needs updating and be able to update with pressing a button. I've see a lot of software in Flatpack and the Ubuntu Software appstore get behind current versions, and this is a way around the issue

probonopd, 2019-09-21:
What would be needed is a Plugin for GNOME Software to manage applications in the AppImage format. I had started writing this some time ago but lost interest in it, anyone is free to pick it up and revive it: https://gitlab.com/probono/gs-plugin-appimage

grigorig, 2020-03-27:
Maybe a Flatpak would be a good idea at this point? I'm not sure if that would be a lot of work, probably not. It might even be possible to just package the AppImage inside a Flatpak.

xarbit, 2020-04-05:
Just came across this ticket while looking if PrusaSlicer exist as Flatpak. Well, I guess not.. 
For the beginning you can simply convert your AppImage as Flatpak and distribute it to Flathub. Its very simple. 
I am willing to do this task and Ill create project for it and if prusa wants, then you can simply take over it any time. :-)

xarbit, 2020-04-09:
@bubnikv you can add me as the volunteer for this task :)

xarbit, 2020-04-10:
The application is submitted to FlatHub and is awaiting review. 
If this is something the Prusa development/release wants to have control over please let me know. 

Feel free to try the flatpak bundle from my repository:
https://github.com/xarbit/com.prusa.PrusaSlicer/releases/tag/v2.2.0-flatpak-testing-1

From my view, now I can finally use my new Fedora Silverblue workstation connected to my MK3s printer :-)

probonopd, 2020-04-10:
> From my view, now I can finally use my new Fedora Silverblue workstation connected to my MK3s printer :-)

Does the AppImage not work for you? If so, which error did you get?

xarbit, 2020-04-10:
Honestly I did not use the AppImage because we want to use flatpak. :) 
I assume it would have worked, it is just not my way to execute software.

kate-shine, 2020-04-15:
> > From my view, now I can finally use my new Fedora Silverblue workstation connected to my MK3s printer :-)
> 
> Does the AppImage not work for you? If so, which error did you get?

I use appimage on Silverblue, but the Flatpak support will be much better. If only for better system integration and automatic updates.

probonopd, 2020-04-15:
Just for completeness, if you want system integration you can use `appimaged` or AppImageLauncher, and for updates the Prusa team could enable update information so that AppImageUpdate can be used.

grigorig, 2020-04-15:
Neither of these are commonly available. Flatpak is a much better choice and already used and installed by default on most, if not all, mainstream Desktop-centic Linux distributions.

kate-shine, 2020-04-15:
> Just for completeness, if you want system integration you can use `appimaged` or AppImageLauncher, and for updates the Prusa team could enable update information so that AppImageUpdate can be used.

Of course I can, but I really don't like mixing packaging systems on my distro. It's messy. And while Flatpak is widely supported and integrated in many distributions, I don't know about any using appimaged :)

Also, one of main reasons for this is convenience: Plenty of users who aren't experienced can find flatpaks in their package manager. 

While Appimage works well, I wouldn't call it convenient or intuitive at all. From the inexperienced user's perspective: 

> 
> I downloaded the slicer, how do I run it?
> [Not as easy](https://i.imgur.com/3eRfFyY.png)
> 
> How do I add it to the system menu? 
> 
> So, It runs, but I still have the "installer" in Downloads, I'll just delete it, what could go wrong?

xarbit, 2020-04-16:
@bubnikv who can I add to have write access to the flathub PrusaSlicer repo once it is created?

probonopd, 2020-04-16:
@grigorig they are available as AppImages.

xarbit, 2020-04-27:
flathub has accepted the PrusaSlicer application, it should arrive in the flathub channel for anyone to use shortly.
 
For the time being I will be the maintainer of this package until upstream (Prusa) is willing to take over this (which should be the goal). 
  
The manifest for this can be found:
https://github.com/flathub/com.prusa3d.PrusaSlicer

Dear Prusa Team, if you need write access and want to take ownership, just open a ticket and FlatHub will take care of this.

I assume this also means this ticket can be closed.

probonopd, 2020-04-28:
> For the time being I will be the maintainer of this package until upstream (Prusa) is willing to take over this (which should be the goal).

While I appreciate your effort, I think cautious users should always download software directly from the official website of the author rather than from any third parties, because only this way one can be 100% sure that the software is delivered unmodified and exactly in the way the author wants it and supports it. This is why I personally avoid downloading applications from anywhere but the original application author's download page.

xarbit, 2020-04-29:
> > For the time being I will be the maintainer of this package until upstream (Prusa) is willing to take over this (which should be the goal).
> 
> 
> 
> While I appreciate your effort, I think cautious users should always download software directly from the official website of the author rather than from any third parties, because only this way one can be 100% sure that the software is delivered unmodified and exactly in the way the author wants it and supports it. This is why I personally avoid downloading applications from anywhere but the original application author's download page.
> 
> 
> 
> I mean, looking at the Flathub repository, it seems like the whole application and its dependencies are built on yet another system (as opposed to Prusa's build system). Who is going to guarantee that the exact same versions of e.g., WxWindows are going to be used, and even if they were, that they are going to be compiled using the exact same compilers etc.?
> 
> 
> 
> It seems like that Flathub approach means that the result are different binaries which are not covered by Prusa's existing end-to-end tests, and hence would require additional support effort.

I do respect you and appreciate your work on AppImage. But seriously? Those arguments are very extreme and pure FUD. I went through the Prusa documentation applied all the suggestions and followed the cmake files and used mostly the same versions and flags. 

Cannot believe this. You know people should use what they want to and I don’t care about politics too much. 
I did this because I needed this and obviously others too.  That is the spirit of open source.

kate-shine, 2020-04-29:
@probonopd so, I guess you don't even use distribution repositories? Because quite obviously, plenty of packages there are linked to libraries with versions different from those used by their developers.

The fact that you prefer appimage is quite obvious from your sealioning, but the fact that it's not a good option for others were explained here.

This issue was marked as "volunteer needed", so what's your problem with @xarbit doing exaclty that?

ivocavalcante, 2020-09-05:
Well, volunteer here. I've created a Snap package for PrusaSlicer, already available here: [PrusaSlicer Snap](https://snapcraft.io/prusa-slicer). It would be great if it was merged upstream, and published under an official Prusa account on Snapcraft.

I volunteer to maintain the package as needed, if allowed. Source is here: [ivocavalcante/PrusaSlicer/snap](https://github.com/ivocavalcante/PrusaSlicer/tree/master/snap).

bubnikv, 2020-09-07:
@xarbit @ivocavalcante 

Thank you guys for your effort packaging our application as FlatPack and Snap. I understand you would like us to take that over. However as of now we don't have the resources in house to do that. As of now (the PrusaSlicer 2.3 and the upcoming PrusaSlicer 2.4) we would be happy, if you could update your packages after our releases.

In the same fashion, there are deb, rpm and aur packages, that we do not directly support. We help the maintainers if they need our help though.
https://src.fedoraproject.org/rpms/prusa-slicer
https://packages.debian.org/sid/main/slic3r-prusa
https://www.archlinux.org/packages/community/x86_64/prusa-slicer/

> > It seems like that Flathub approach means that the result are different binaries which are not covered by Prusa's existing end-to-end tests, and hence would require additional support effort.

> I do respect you and appreciate your work on AppImage. But seriously? Those arguments are very extreme and pure FUD. I went through the Prusa documentation applied all the suggestions and followed the cmake files and used mostly the same versions and flags.

I did not review the FlatPack by @xarbit. AFAIK the FlatPack is supposed to support the concept of shared libraries packed as FlatPack. I suppose this is what @probonopd mentioned when he expressed his concerns about the binary produced by you to work as well as the binary produced by us. We use a particular wxWidgets library and we do patch the wxWidgets library if needed to patch various quirks and improve performance. 

We (Prusa Research) could only guarantee the correctness of our builds only and we really appreciate the simplicity of the AppImage and that AppImage does not require any other dependence. I did not use Linux on desktop for a long time myself. I installed Ubuntu on my company desktop couple of months ago to use it as a daily driver and I think that by default the flatpack server is not there as it is a competing solution to their own snap. Now we have many users who are not very technical and they have no idea what a flatpack is. For them the AppImage is likely a simpler solution then to study and install a flatpack support system on their Ubuntu. I don't claim I am correct as I am not a seasoned Linux desktop user though, take it as an impression of an average non-technical Linux user.

xarbit, 2020-09-07:
@bubnikv thanks for your input. I will update the flatpak in the upcoming days.

ivocavalcante, 2020-09-07:
Ok @bubnikv , very understandable. I'll keep updating Snap as well.

mrjohnc, 2021-01-12:
> @bubnikv thanks for your input. I will update the flatpak in the upcoming days.

I noticed something interesting, I installed PrusaSlicer on my laptop using Flathub, when I open Prusa Slicer it says 2.3 is available but then Flathub say 2.2 is the latest version.

mrjohnc, 2021-01-12:
> 

One exception to this is Chromebooks, when Flatpak and snap images are much easier to use than app images (which require you to understand the command line interface and change the commands based on the app image name. Is there any way to know how many people are using the snap or flathub image?

xarbit, 2021-01-12:
Unfortunately, due to life events I am currently not able to update the flatpak Package on flathub that fast. I still have it on my list.
Please be patient or feel free to open merge request against the flathub PrusaSlicer package and I'll happily look at it and merge.

mrjohnc, 2021-01-12:
> Unfortunately, due to life events I am currently not able to update the flatpak Package on flathub that fast. I still have it on my list.
> Please be patient or feel free to open merge request against the flathub PrusaSlicer package and I'll happily look at it and merge.

Sorry, that was not meant to be an ask to update it, just to note that it can know when there's an update available even if its not on Flathub yet

NOVATechnocrat, 2021-05-11:
I find that people quite often advocate the use of technologies they don't understand. The usage of flatpak and snap being two of them. 

If a person has never performed the following:

1) $ `flatpak update` (and "`flatpak uninstall --unused`" for good measure)
2) $ `sudo snap refresh`

Then they permitted their system to go un-patched worse-than-Microsoft-style.

With that said, I'd like to offer a note to all:

If you install any flatpak/snap, including those listed here, you are fully trusting these maintainers to stay on top of packaging these applications. Just because Prusa updates PS, it doesn't mean the Flatpak or snap runtime you installed will be up to date. Furthermore, the Flatpak and snap runtime is updated hyper-frequently, meaning the maintainers need to repackage it when Prusa updates their package or when ANY single library they packaged it with are updated which is easily once a week.

I packaged Linux Discord and Twitch applications wrapped in Electron and my Twitch application is out of date. Since I wrapped it using Electron, I and my users are beholden to any Electron/Chromium bugs/vulnerabilities so I need to repackage it now.

But that's part of it.

I thank these maintainers for their effort and time. I just see many users, particularly those new to Linux being pushed in to Flatpak/snap environments without even knowing how to update their system. If you ran "flatpak update" and it returned anything other than "Nothing to do." then you are a noob and you just got schooled. If this is the case, think back at how many times you sung flatpaks' praises and recommended it without even knowing how to update it.

With that said, it's always going to be best to get the appimage directly from (wherever) Prusa, for security and other reasons. However, if you want the Flatpak or snap and understand the shit that goes along with it, then please -by all means- install the apps these maintainers have graciously packaged for us.

Sorry, this was largely a rant.

kate-shine, 2021-05-12:
@NOVATechnocrat Please, do tell me, how is not running a `flatpak update` different from not running `apt upgrade`, `dnf upgrade`, `pacman -Syu` or any other package manager update? 

>If you ran "flatpak update" and it returned anything other than "Nothing to do." then you are a noob and you just got schooled.

No, that just means that I am currently updating my apps.

> it's always going to be best to get the appimage directly from (wherever) Prusa, for security and other reasons.

After you pointed out the necessity to update, your proposed solution is to use the release channel that doesn't allow for simple updating and doesn't allow the update of runtime environment independently from the packaged app. Amazing.

NOVATechnocrat, 2021-05-14:
Not running flatpak update is the exact same as not running apt upgrade. You need to update all of them, On a modern Linux system you need to update flatpak, apt, and snap (among others) and missing a single one of them is putting your system at risk.

If you just ran that for the first time, then you are an utter noob, It's the same as you just finding out how to run sudo apt upgrade or the equivalent. Doesn't matter how many "years you've used Linux", if you are just finding out how to update your system, you have been a liability that entire time. That person, -the one just now running it- is the person I'm targeting and since you seem upset about my comment then I'll assume you just now updated flatpak after x years? 

Of course, if you know how to update flatpak and have been doing so this entire time, I'm not talking about you.

I expect you will see many people going years, not knowing they need to update it. And now that I said that, I suspect you will see those people advocating technology they don't understand. That's all I'm getting at.

`After you pointed out the necessity to update, your proposed solution is to use the release channel that doesn't allow for simple updating and doesn't allow the update of runtime environment independently from the packaged app. Amazing.`

Of course I recommend you to use the official release channel. I'm not sure you are aware of the process a package maintainer needs to go through to update the package. That PS flatpak doesn't automatically update itself. This kind soul that offered to do so must:

- Acknowledge Prusas official release (through it's official release channel,, ie: the appimage) has been updated.
- Download that appimage or source in to their build folder.
- Use whatever library they used to build or convert the appimage in to the flatpakl/snap.

It doesn't automatically update itself and I'm not sure you understand that.

All of these things could take an indefinite amount of time to complete. 

You are right about the appimage though, you must check periodically for updates. However, when you install a flatpak, the MAINTAINER must check the official channel periodically which displaces the responsibility on another person that you don't know.

Let's take a quick look at the "unofficial" version (flatpak) and compare it to the official one:

First we will note the difference in OpenGL versions, which may explain why there is a different tone in the gray color. I'm unsure of why they are different, however you'll note that the unofficial one uses a considerably older library. What that means is this package is NOT_UP_TO_DATE or at least, older than the official version:

[image]

Then note the difference in gray:

[image]

Finally notice the difference in build version:

[image]

Yes, use the official version (100% of the time, all down a line) and periodically check for updates. Again, I thank the package maintainer for their time and effort.

kate-shine, 2021-05-14:
>On a modern Linux system you need to update flatpak, apt, and snap (among others) and missing a single one of them is putting your system at risk.

On a modern Linux system, you usually have some update manager, that does that for you, including snap and flatpak. If you don't have one, you are probably using some more DIY distro like Arch, and then you should be experience enough to handle it yourself.

>Of course I recommend you to use the official release channel. I'm not sure you are aware of the process a package maintainer needs to go through to update the package. That PS flatpak doesn't automatically update itself. This kind soul that offered to do so must:

You are literally answering under the issue of people asking upstream to start maintaining flatpaks as well. So, your objections about the unofficial package are quite offtopic. Yes, it's slightly outdated, but what point are you exactly trying to make here in this thread?

mrjohnc, 2021-05-14:
I'm unsubscribing due to toxicity, hope this gets completed but not going to stay

NOVATechnocrat, 2021-05-14:
I have no objections of the unofficial package, I think it's great that they took up that project. I think it would be great for Prusa to package the slicer in Flatpak among every single other platform/runtime imaginable.

But you know how to update Flatpak now, right?

kate-shine, 2021-05-15:
@NOVATechnocrat I am maintaining an official flatpak build of one app. So, what do you think?

But obviously, you did have nothing to say to the topic of this post and just went here trolling, so whatever.

NOVATechnocrat, 2021-05-16:
> @NOVATechnocrat I am maintaining an official flatpak build of one app. So, what do you think?
> 
> But obviously, you did have nothing to say to the topic of this issue and just went here trolling, so whatever.

All I was saying is that many people advocate technologies they don't understand. You are not disputing that. You are flaming me because I pointed it out. I mentioned a tell-tale sign of a person that does such a thing, is one that doesn't know how to update flatpak.

What would you call a person that -after using flatpak for years- just realizes you have to manually update it? I call that person an utter noob, a liability to any network that person is on, a person breaking the number one rule of security which is failing to UPDATE a system.

I mean it in a good way. A "get it together, TIGHTEN UP" type of way.

Why are YOU taking offense to that? I don't want to insinuate you are just finding that out, but you are acting like it.

I'm not trolling. After reading every comment here, I am certain there are misconceptions about flatpak technology. Such as that flatpak provides some automagic update mechanism. If Prusa were to release an update, nothing would happen at all. You would NEVER receive an update notice through flatpak. 

Until when? At what time would you receive a notification to update the package? When does this notification to update appear?

> After you pointed out the necessity to update, your proposed solution is to use the release channel that doesn't allow for simple updating and doesn't allow the update of runtime environment independently from the packaged app. Amazing.

The best way to have an up-to-date Prusa Slicer application is to keep checking yourself, everyday, sign-up for their newsletter, their RSS feed, whatever you have to do to be alerted that they updated it, and then download it yourself. This is because Prusa (it appears) doesn't natively offer flatpak support. They offer appimage. I'm not saying appimage or flatpak is better than one another. I'm saying Prusa supports appimage and because of that, you are best served getting it from their release channel: **Full Stop.**

You are putting 100% faith that the maintainer will check for updates every day? How often is a suitable response time for a package maintainer? If PS releases an update and the package maintainer doesn't get around to it for a month, meaning you don't receive a notification in flatpak that there is an update available, is that suitable for you? I has to be. The maintainer is doing it from their own good graces and you have no control over it -at all-.

With that being said, in order to have the most up-to-date (and most secure) appimage possible, you will check Prusas site as often as possible and download it directly from there. Unless you have an argument to that, please leave me alone and no I'm not trolling you.

You flamed a guy and said he was "sealioning" (laff) when he made an entirely coherent and accurate statement:

> While I appreciate your effort, I think cautious users should always download software directly from the official website of the author rather than from any third parties, because only this way one [...]

The only valid argument against that comment is: "I just wish Prusa would support flatpak, rather than appimage."

Well, they don't -yet- and as this person points out:

> "cautious users should always download software dirrectly from the official website".

You all started arguing THAT and THAT is why I interjected myself in to your argument. If you argue that, then you are wrong and I will not permit you to derail this topic with your hurt feelings without interjecting. Any person that states otherwise should be challenged and be removed of positions of important decision making.

I support the original posters request for Prusa to support flatpak.

bubnikv, 2021-08-30:
@kate-shine @xarbit

FYI
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md#desktop-integration-prusaslicer-24-and-newer

bubnikv, 2021-08-30:
I have created a Wiki page with the 3rd party provided builds.
https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-on-Linux---binary-distributions

Also the differences in UI look raised by @NOVATechnocrat  may be due to a different version of wxWidgets being used by our static builds vs. 3rd party builds.

As wxWidgets 3.1.xx is still considered unstable (binary API is not considered stable), Linux distros often contain old wxWidgets 3.0 only. While PrusaSlicer compiles against wxWidgets 3.0, one may experience some user interface glitches.

ivocavalcante, 2021-08-30:
> @kate-shine @xarbit
> 
> FYI
> https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md#desktop-integration-prusaslicer-24-and-newer
> 
> If PrusaSlicer is compiled with SLIC3R_FHS enabled, then a desktop integration support will not be integrated. One may want to disable desktop integration by running
> 
> cmake .. -DSLIC3R_DESKTOP_INTEGRATION=0
> 
> when building PrusaSlicer for flatpack or snap, where the desktop integration is performed by the installer.

Thanks for pointing out, @bubnikv . Will check on next Snap build.

xarbit, 2021-09-02:
@bubnikv thanks .. the flatpak for the upcoming 2.4 is being reworked on and will use the wxwidgets and other dependencies provided by prusa3d and not like before from the flatpak repo.  

Its already being built and made available in the beta branch and will be tested.

xarbit, 2021-09-02:
Here is the first working PrusaSlicer 2.4-alpha1 as flatpak which uses the dependencies provided by Prusa3d. 
The new build also adds the separate Prusa G-Code Viewer and adds Desktop Action (right-click) to open G-Code Viewer.  

Please give it a shot and let me know what you think:

`flatpak install https://dl.flathub.org/build-repo/56118/com.prusa3d.PrusaSlicer.flatpakref`

[image]

@bubnikv once https://github.com/prusa3d/PrusaSlicer/pull/6875 is merged, ill remove the desktop files from flathub repo.

bubnikv, 2021-09-02:
> @bubnikv once #6875 is merged, ill remove the desktop files from flathub repo.

merged

> The new build also adds the separate Prusa G-Code Viewer and adds Desktop Action (right-click) to open G-Code Viewer.

That is cool. We may want to update our manual desktop integration in the same fashion. We wonder whether this kind of integration will work on Chrome OS as well, two Desktop files pointing to a single binary did not work.

xarbit, 2021-09-02:
@bubnikv does ChromeOS follow the feedesktop.org specifications for .desktop files? 
If so, then I don't see why it wouldn't work.. Unfortunately I don't have ChromeBook/ChromeOS to test. 

https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html

Feel free to look in the desktop file I provided. You might be able to use Desktop Actions. 
https://github.com/prusa3d/PrusaSlicer/blob/c232decb2a37a966da055ef55d402782c04c97a7/src/platform/unix/flatpak/com.prusa3d.PrusaSlicer.desktop#L16

ivocavalcante, 2021-09-02:
Hi @xarbit and @bubnikv ,

I see the code contains desktop files specific for Flatpak; I really know nothing about Flatpak build system, but wouldn't it be possible to have a single set of desktop files on the code, so Prusa official versions and also Snap package could benefit? I mean, having GCode viewer action is a really cool addition and I plan to bring it to Snap, but since it was added to Flatpak's file version (and I use a buildtime modified version of the "official"), Snap won't get it by default (I can make it get it, but that's not the point). Also, you're including translations, which is also really cool, but again: duplicate files. Is there a way to join these two, so we can all benefit (having actions and adding translations in a single point) ?

I'm  not sure this is the right place to post this discussion, but since you're all here, I took the chance. Feel free to move it somewhere else, if you see fit.   :)

xarbit, 2021-09-03:
@ivocavalcante yes, I agree .. there is a way, even if I have to use "sed" ..  I need to do some restructuring and testing to make that possible. But it's doable.

ivocavalcante, 2021-09-05:
> Let me try ..

Great, Jason. Let me know if you are successful, so I started adjusting things here too. Thanks!

xarbit, 2021-09-05:
@ivocavalcante 
yes, I am done implementing on my side and everything is shaping up pretty good. 
When you are done adjusting things on you're side, let me know as well so I can remove this part:

https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/1daf65a3590d799806d079a74ab462019fd42f51/com.prusa3d.PrusaSlicer.yml#L227

ivocavalcante, 2021-09-11:
Great Jason, I believe I'm all set here. Waiting for changes to be merged.

ivocavalcante, 2021-09-28:
Well, there it is, @xarbit and @bubnikv . See if it's up to the task.
#7023

xarbit, 2021-09-28:
@ivocavalcante https://github.com/prusa3d/PrusaSlicer/pull/6879 was already created ;-)

ivocavalcante, 2021-09-28:
Great! Would you, please, include pt_BR translation I added on my fork? Or you prefer I open a PR to your repo? I'll close my PR.

xarbit, 2021-09-28:
@ivocavalcante please if you don't mind, PR to my branch : merge-desktop-files 

Thanks

ivocavalcante, 2021-09-28:
Done!

trainmeditations, 2024-06-30:
@xarbit I'm having issues with the appimages on Debian so I thought I'd check out the flatpak. The wiki said you managed it and to post issues here. I'm having a problem with it though where if I try to launch it from a file nothing seems to happen. I tried opening it from the terminal with xdg-open and got the following

'''
kf.service.services: KApplicationTrader: mimeType "x-scheme-handler/file" not found
--------------------------------------------------------------------------
Message: 21:57:39: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
Message: 21:57:39: INFO: using dark theme variant
No such file: file:///run/user/1000/doc/e7d834e3/3-each-100-75-50.3mf
'''
The file is in that location when I look, the error seems to be because it's using a file:// path

Seems it might be related that I'm on KDE Plasma. I'm not overly familiar with flatpak but I'm trying to find out of there's some element flatpak is missing to make this work. Would be great for your insight though.

xarbit, 2024-06-30:
> @xarbit I'm having issues with the appimages on Debian so I thought I'd check out the flatpak. The wiki said you managed it and to post issues here. I'm having a problem with it though where if I [...]

Hi @trainmeditations 

thanks ...
The best would be if you could please open an issue here: 

- https://github.com/flathub/com.prusa3d.PrusaSlicer/issues 

we will look in to it .. 

Thanks

Jason

trainmeditations, 2024-06-30:
> > @xarbit I'm having issues with the appimages on Debian so I thought I'd check out the flatpak. The wiki said you managed it and to post issues here. I'm having a problem with it though where if I try to launch it from a file nothing seems to happen. I tried opening it from the terminal with xdg-open and got the following
> > 
> > ```
> > kf.service.services: KApplicationTrader: mimeType "x-scheme-handler/file" not found
> > --------------------------------------------------------------------------
> > Message: 21:57:39: Starting PrusaSlicer flatpak with entrypoint script
> > --------------------------------------------------------------------------
> > Message: 21:57:39: INFO: using dark theme variant
> > No such file: file:///run/user/1000/doc/e7d834e3/3-each-100-75-50.3mf
> > ```
> > 
> > The file is in that location when I look, the error seems to be because it's using a file:// path
> > 
> > Seems it might be related that I'm on KDE Plasma. I'm not overly familiar with flatpak but I'm trying to find out of there's some element flatpak is missing to make this work. Would be great for your insight though.
> 
> Hi @trainmeditations 
> 
> thanks ...
> The best would be if you could please open an issue here: 
> 
> - https://github.com/flathub/com.prusa3d.PrusaSlicer/issues 
> 
> we will look in to it .. 
> 
> It would also be helpful if you would paste the commands that you tried to do. 
> 
> Thanks
> 
> Jason
> 
> 
> 

Done

[3 more replies]

closed 2024-11-21 by lukasmatena