prusa3d/PrusaSlicer #1125: Features Request for Slic3R

tonysmartantzoulatos, 2018-08-05:
Hello Prusa Reasearch,

I was told by tech support to type an email with slic3r requests and then i was responded by Jennes De Schutter that i should type them here. Here it goes then.
a) To be able to set an unlimited amount of extruder temperatures for individual layers heights. I am not asking for BED temp only extruder. Since a feature like this makes it easy to print hard materials that accept very little cooling and in most cases almost none that happen to have a large base and alot smaller top that might also have an amount of overhang that with this feature you can print WITHOUT the use of supports. 
b) Multiple steps of cooling for different layers heights. For filaments that use cooling and you might have variable geometry dimensions, a feature like this can keeps the filament performing as it should. In the existing version you can only disable for some layers.
c)In the existing version and as owner of MMU i love the soluble support feature but its mainly designed for use of PVA with PLA. There are way and other filament to combine but none will give you the PVA look. Suggestion would be to have the Soluble interface feature work with other filaments also. As an example i will say that i have tried PVA with other filaments but could not build the interface layer and my suggestion would be to have 2/3 interface layers of you existing filament that builds the majority of supports on which PVA can be build on. This way it might be possible to fill the gap with PVA that can even be printed at temps close to ABS.
d) Based on the "request" above there is also the mater of purge per filament in order to have clean filament after each change. At the moment as the system is set up on Slic3rPE is not really understandable, unless there can be some clarification in how it work. My suggestion would be a 3rd option that takes about mm (pure mm)of filament being extruded on each change.

Thank you for your time and hope my ideas/reguests make some sense.I 
### Version
_Version of Slic3r used goes here_1.40.1

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-08-07:
When entering issues or requests, please enter them one by one, and use an expressive title, so we can sort the issues out.

c) You are right, it would be nice to have a PLA support interface layer before a PVA support interface layer.

d) It is a matter of personal preference. The purge values are entered in cubic mm, which is linearly dependent on filament length (1.7mm dmr filament has 2.405mm filament cross section area). So you just need to divide your filament lengths with 2.4 before entering them as cubic mm.

hexane360, 2018-08-09:
a) and b) are already possible with current Prusa Slic3r, using the G-code macro language.

[Here's](https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language) a full write-up. The gist is that you put a conditional expression in "Printer Settings"->"Custom G-code"->"Before layer change G-code", which runs different commands based on a value of `layer_z` or `layer_num` (zero-indexed FYI). There's an example of this in action [here](https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language#continuous-decrease-of-nozzle-temperature).

To know which g-codes to put inside the conditionals, you can refer to the [Marlin firmware wiki](http://marlinfw.org/meta/gcode/). Slic3r doesn't care what you put in here, so just make sure it works on your printer. Here's a couple useful ones:
```
M106 S[0-255]  ;set fan to speed 0-255
M104 S[T]      ;set hotend to temp T
M109 S[T]      ;set hotend and wait for heating
M109 R[T]      ;set hotend and wait for heating or cooling
M140 S[T}      ;set bed to temp T
M190 S[T]      ;set bed and wait for heating
M190 R[T]      ;set bed and wait for heating or cooling
```

tonysmartantzoulatos, 2018-08-12:
@hexane360 apologies, since i am not so much into code. I am more on the design of usable parts side.
i will test the suggested way, but my theory was to be able to have something easy that does not require i know how to type marlin code into my custom g-code window and create a "profile" for a specific print.
In order to understand properly....if i want changes on layers 50,100, 150 and 200 i must type.
;[50] (50 being layer_z)
M104 S[T]      ;set hotend to temp T
;[100]
M104 S[T]      ;set hotend to temp T
;[150]
M104 S[T]      ;set hotend to temp T
;[200]
M104 S[T]      ;set hotend to temp T

Since i dont need extruder to wait for the change since i will simply be lowering as i go higher.

hexane360, 2018-08-13:
You're almost there. Those are the correct commands, but you just need to have Slic3r's conditional statements around. Again, [this](https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language#continuous-decrease-of-nozzle-temperature) is a working implementation. In your case, it would look like this:
```
{if layer_num == 50}M104 S155
{elsif layer_num == 100}M104 S150
{elsif layer_num == 150}M104 S200
{elsif layer_num == 200}M104 S200
{endif}
```
The curly braces tell Slic3r to interpret the inside as a command to Slic3r instead of the printer. "if" includes the enclosed text if the condition inside is true, "elsif" is short for "else if", and includes the enclosed text if it is the first true condition in the if statement. "endif" indicates the end of the if statement.

Note that layer_num is "0" for the first layer.

tonysmartantzoulatos, 2018-08-14:
@hexane360 Thank you for your assiastance.
I was able to create such a profile and testing it with some temp towers to see how it works.

closed 2018-08-14 by tonysmartantzoulatos