prusa3d/PrusaSlicer #1126: End Code and Start Code Improvements

GreatGrizzly, 2018-08-06:
Frankly, the current start gcode and end gcode suck. The start gcode causes plastic to get all over the bed, attach to the nozzle, and is just inefficient time wise for starting prints.

The end gcode is also pretty bad, leaving no room to clean nozzles, or letting plastics bend up and stick to the nozzle when they cool (Very prevalent with PETG and TPU). 

Here is the following GCode I use that minimizes mess, eases cleaning, and streamlines the the printer start state and end states. I think something similar to the following should be part of the default settings. Input is welcome to try to get this as generic as possible for most printer setups:

**Start GCode**
 M115 U3.3.1 ; tell printer latest fw version
M83  ; extruder relative mode
M140 S[first_layer_bed_temperature] ; set bed temp, continue on
M104 S180 ; set extruder temp to a value below what most filaments will use to prevent leaking, TPU leaks on anything higher than 180
G28 W ; home all without mesh bed level
G1 Z150 X125 Y-2 F4800 ; get head above bed to keep PINDA probe cool, move the nozzle up front and away from left side camera to make cleaning easier, Set feed rate to do this as fast as possible.
M190 S[first_layer_bed_temperature] ; wait for bed temp.
G28 Z ; home z without mesh level
G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp to first layer temp, continue on
G1 Y-3.0 F1000.0 ; go outside print area
M109 S[first_layer_temperature]; wait for extruder temp
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0

**End GCode**
G4 ; wait for buffer to clear
G92 E0; set extruder value back to 0
G1 E-2 F120; retract 1mm at speed of 120mm/s
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X125 Y210 F4800; Bring print surface forward as fast as possible per min, center x to try to keep filament on spool and hotend cable off usb plug
G1 E10 F1000; Crank out some plastic to weigh down the plastic thread.
M104 S0 ; turn off hotend
G1 Z205 F1200; Raise to top.
M84 ; disable motors

GreatGrizzly, 2018-08-07:
@mloidl Perfect! Just what I was looking for. Anything to save time when getting the printer ready to go is a good addition. Updated the original code.

bubnikv, 2018-08-07:
@jindrichbenes is looking into it

ayberkozgur, 2018-08-14:
@GreatGrizzly I like your improvements a lot in theory, I'll try them out together with some of my improvements surely soon.

The goal is of course to eliminate any and all ooze in a fully automated manner before and after each print, but I don't imagine that is practically possible even with any amount of retraction. However, the next best goal, that is to make cleaning predictable (which in turn eliminates any and all ooze not in a fully automated manner but with manual intervention), is very much doable as you have done.

I've found the intro line to be quite effective in cancelling previous retractions, so I'd be much more liberal with retractions at the end of prints to save a bit of filament and make cleaning easier. Namely, I'd replace:

> G1 E10 F900; Crank out some plastic to weigh down the plastic thread. This prevents material from curling up as it cools, sticking to hotend.

With something like:

> G1 E10 F900; Crank out some plastic to weigh down the plastic thread. This prevents material from curling up as it cools, sticking to hotend.
> G1 E-20 F2000; Retract to limit further oozing

The exact feedrate and amount to ensure that the intro line succeeds should of course be tested.

ayberkozgur, 2018-08-14:
> The reason I am not retracting so much is I don't want to clog the hot end
by pulling plastic into the top where it can't be reheated.
Fair enough, I see your point. Having just tried your start gcode, I can say that waiting for full temperature at (0,-3,0) effectively traps the oozed filament between the nozzle and the bed, which does not get dragged along when the intro line starts. In the end you're left with a totally clean nozzle at start of the layer 0 skirt.

GreatGrizzly, 2018-08-14:
Thanks for testing this. In older iterations of the end code I pushed the nozzle against the bed to "plug" it. This prevented leaking... but caused the nozzle to stick to the bed.

GreatGrizzly, 2018-08-14:
@bubnikv What is the parameter to get the bed X and Y size values? The tooltip says its rect_size for both values.

ayberkozgur, 2018-08-15:
Now, having tried out your end gcode, I see two potential problems with it:
1. Raising Z until 205 is fine at the end of the print, but as soon as the next print starts, home all starts with homing X and untangles one loop of filament from the spool on the stock spool holder. This happens every time, and on spools that are not even brand new. We could raise Z until something like 150 instead, or we could start by homing Z, then X and Y (will this destroy the Z homing somehow? I don't know yet).
2. `G1 E10 F900` seems too much (although I'm still in the process of testing it), the extra bit to weigh down is actually a 40-50mm long line and touches the bed. This causes that long line to incline on one side, also pulling the bit that's still on the nozzle to that side as well. It's a minor issue but I'm thinking of something more like `G1 E5 F900` instead. I'll try and report.

GreatGrizzly, 2018-08-15:
1a. That makes sense. I never had unraveling with my spools though. But Z change should definitely  be first, to tighten the roll before X/Y movements.

1b. The reason I put Z205 is I am assuming I cant go any lower because something that was printed might be at the maximum Z height of the printer. The chances that someone uses the full z axis for a print is rare...but its a possibility.  As for wear on the motors...I haven't considered that. I don't know how significant this would be as the MK3 is built to skip steps for homing procedures.

I am not sure of the logic for why z homing raises Z before lowering. The only thing I could think of is if the nozzle is below zero when it starts. But if thats the case, the probe will be on...indicating that its too close to the bed. This could be a potential bug?

GreatGrizzly, 2018-08-16:
After some testing, it appears that the Z axis cant be homed before X and Y. This is built into the printer firmware itself. The parameter is Z_SAFE_HOMING. So we are stuck with my current implementation.

ayberkozgur, 2018-08-16:
I'm not sure either of the logic behind the initial Z raise when homing; it could make sense from the perspective of clearing the extruder from any potential object on the bed surface, but I'm guessing it's supposed and assumed to be totally clear before each print.

For now I'll go with something like `G1 Z{max(layer_z, 150)} F1200` at the end of prints as I don't have the requirement to push unparameterized gcode from somewhere else. The exact amount is to be determined by whether any filament unravels from the spool at the start of the next print.

Panayiotis-git, 2018-08-16:
Regarding the start g-code, at the point where you move the head at 0,-3,0 in order to trap any oozed filament between the bed and the nozzle.
I've used this technique on a MK2 for about 9 months. It worked. But then the PEI has got unstuck from the bead at this corner and later it broke.

I don't now if this is still possible on a MK3 flexible steel plate.

ayberkozgur, 2018-08-16:
Very interesting point. Maybe we could offset the waiting point and intro line by something like +50 on the x axis to not stress the corner too much?

ayberkozgur, 2018-08-18:
@Panayiotis-git I've offset the ooze trap and intro line X coordinates by +75, my intro gcode looks like this:

> M115 U3.3.1 ; tell printer latest fw version
> M83 ; extruder relative mode
> M140 S[first_layer_bed_temperature] ; set bed temp, continue on
> M104 S{first_layer_temperature[0]-70} ; set extruder temp to 70° less than the target temp to prevent oozing, continue on.
> G28 W ; home all without mesh bed level
> G1 Z150 X125 Y-2 F{machine_max_feedrate_y[0]*60} ; get head above bed to keep PINDA probe cool (needed for MK2), move nozzle up and away from left side camera to make cleaning easier. I don't want to raise all the way because this might cause new filament spools to come undone which can tangle.
> M190 S[first_layer_bed_temperature] ; wait for bed temp.
> G28 Z ; home z without mesh level again, as bed temp has changed
> G80 ; mesh bed leveling
> M104 S[first_layer_temperature] ; set extruder temp to first layer temp, continue on
> G1 X75.0 Y-3.0 F{machine_max_feedrate_y[0]*60} ; go outside print area
> M109 S[first_layer_temperature]; wait for extruder temp
> G92 E0
> G1 X135.0 E9.0 F{machine_max_feedrate_e[0]*9}; intro line at max mm/s
> G1 X175.0 E12.5 F{machine_max_feedrate_e[0]*9}; intro line at max mm/s
> G92 E0
> M221 S{if layer_height==0.05}100{else}95{endif}

Although still close to the edge, I think this location would provide relatively better heat dissipation compared to the corner, extending the PEI coating lifetime.

Otherwise, I've found this intro to be quite robust so far. As long as the nozzle is clean from the previous print (which can be done easily when it's cold thanks to the end code) I can simply start the new print and walk away. No preheating, no ooze cleaning in real time. I'd like to see at least this performance in the official start/end gcode.

tjcolejr, 2018-09-03:
I found that heating the pinda probe itself to a temp higher than it would usually ever be when starting a print reduced the z height variance more than probing a hot bed with a pinda that may, or may not be warm already.  For instance, when you start a new print shortly after finishing one, the pinda is warm.  When starting a new print on a dead cold printer, the pinda is cold.  This DOES make a difference.  So before mesh bed leveling, I move the nozzle to bed center and then heat the bed to 60.   I also preheat the nozzle to 130.  So mesh leveling is done with consistent hot temps.  I do not have to change my z height when changing filament types UNLESS a specific filament prefers to be squished or not squished on the first layer.

jindrichbenes, 2018-09-03:
First of all, we were unable to replicate the issue, during testing, the nozzles practically didn't leave ooze on bed. Secondly, letting nozzle cool down and heat up again added some 75 seconds to start print time, That can be unpleasant to some users, to which, mesh bed leveling takes too long already as it is. Third issue would be, that should there be any leftover  plastic on cooled nozzle, this plactic could compromise the mesh bed leveling. This does not happen with hot nozzle.

Still, we will continue looking into this issue to come up with a solution.

ayberkozgur, 2018-09-03:
@tjcolejr I did not test this but it does make sense to "preheat" to known temperatures and calibrate/level with those.

What's more, I realized that `first_layer_temperature[0]-70` is not completely enough to keep PETG from oozing (160 degrees), there's just a little bit that comes out but still there is ooze. I think a global 140 (or maybe 150 degrees, needs to be tested) would be better in the end.

tjcolejr, 2018-09-03:
The oozing during mesh bed leveling using the standard gcode is simply not debatable.  It has been a complaint for at least 2 years.  And if the variance of the pinda's accuracy due to temperature wasn't an issue, then the pinda 2 would not have a temperature sensor.  Heating the pinda and bed to a consistent temperature well above ambient prior to probing eliminates the temperature variable.

@jindrichbenes<https://github.com/jindrichbenes>:

Seriously, I find it very hard to believe that you didn't have ooze during the initial heat up at the start of the print, which is a normal phenomenon unless measures are taken against it. Without manual cleaning during the limited time when the print is about to start (while the nozzle is hot BTW), this gets on the bed and the first few layers very easily. What's worse, it tends to curl up and get on the nozzle, polluting it over time. This is true for all the filaments I tested so far, including the Prusa filament that came with the printer. If you really need, I can provide a video of exactly this happening.

The start routine does not cool down the nozzle in any way, the temperature deltas are the same, and the time should be very similar. If you're referring to the cooling after the print ends, it will happen enough in any case while the object(s) is(are) removed from the bed and the intro line is cleaned. But seriously, given that FDM prints still usually take on the order of hours, if 1-2 minutes extra (which ensures a clean first layer) bothers a user, they should probably give up 3D printing.

Mesh bed leveling is not done with a cold nozzle, 140+ degrees is still enough for the filaments to be quite compliant. In any case, the nozzle should be cleaned after the print is done and the nozzle cools down, and the ooze stops completely (which easily happens until the object(s) is(are) removed from the bed and it is cleaned). If the nozzle is cold to the touch, it can be easily done by hand as well.

I'm honestly suspecting that you're testing other gcodes than the ones given in this issue...

joeynuggetz, 2018-09-03:
Tried running this code on my MK2 and latest slic3r won't export after slicing. 
Failed to process the custom G-code template start_gcode
Parsing error at line 4: Invalid utf8 sequence
M104 S{first_layer_temperature[0]-70} ; 

I guess that's not available at the firmware level that us MK2 users are stuck with.

bubnikv, 2018-09-03:
>  Tried running this code on my MK2 and latest slic3r won't export after
slicing.

By my mistake, you received Slic3r profiles incompatible with your current
Slic3r. Sorry about that. The best for you is to update your slic3r:
https://github.com/prusa3d/Slic3r/releases

joeynuggetz, 2018-09-03:
I'm already running 1.41.0.

GreatGrizzly, 2018-09-16:
@joeynuggetz Try manually typing in the line that is giving you trouble instead of copying and pasting.

hawalter, 2018-10-01:
I work mostly with PETG on my mk3 which lives up to its reputation of oozing and stringing. I always have oozing and stringing during leveling. Two observations that I hope can be fitted into the above discussion.
First, every print leaves a string from the end of the print to the home position so please consider a retraction at the end of the print.
Second, almost every print leaves a string from the end of the first line to the start of the print. Again, a retraction should help.

mwhincup, 2018-10-01:
Shouldn't this issue be in the Slic3r Settings section?
https://github.com/prusa3d/Slic3r-settings

GreatGrizzly, 2018-10-06:
@mwhincup I have an issue open there as well. Issue  # 45

I am getting a lot more traction in this area though.

hawalter, 2018-10-07:
@GreatGrizzly I am new to 3D printing, slic3r and Gcodes so I will have to use english instead of gcodes. 
In slic3r->Printer Settings->Extruder 1 under the Retraction section, there are numerous retraction parameters. In short, apply them every time after filament extrusion and before a move.

In particular, retract 
   at the end of the first strip and before the move to either the brim or the build.
   at the end of the brim (if there is one) and before the move to start the build.
   at the end of the build and before the move to home.
I am wondering (hoping!) if the retraction before the home will reduce the pressure in the extruder and cause less oozing for the next bed leveling.

@jindrichbenes You said you could not replicate the oozing. Did you try PETG?

ayberkozgur, 2018-10-07:
I also realized that enabling the fan at 100% at the end of the print to speed up cooling might be a good way to reduce oozing by getting to <150C as fast as possible.

GreatGrizzly, 2018-10-24:
@hawalter  I am not quite following you. It might be better for you to post some gcode to explain what you mean.

@ayberkozgur That sounds like a good idea, though I wonder if setting the fan to full speed might have the plastic swing up and stick to the hotend.

ayberkozgur, 2018-10-27:
That's a possibility but I doubt the fan would be powerful enough. I'm 
planning to test it soon.

ayberkozgur, 2019-02-17:
Is there any planned work on this issue?

apphead, 2019-12-11:
@GreatGrizzly I ended up googling for _rect_size slic3r_ like me: have you found the answer? Anyone?
I searched the code and tried x_max, y_max, rect_size[0], rect_size.x but to no avail

wolfgangmauer, 2020-07-22:
@apphead Did you find a solution? I am also interested in it.

ingotero, 2020-10-09:
Hi everyone, after a lot of trial and error I finally found what I call "The Holy Grail" of the start and end G-Codes! lol ;p
I extracted some ideas of this post (thanks for that!).
It is clean after the mesh bed leveling and clean after the end of the printing.

I used PETG for the testings.
My printer is a Prusa MK2.5S

I found that the secuential retractions are better than the continuous retractions for this purpose (having the nozzle 100% clean).

**Using 230° C**

Start G-Code:

M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.9.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83  ; extruder relative mode
M104 S160 ; set extruder leveling temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S160 ; wait for extruder leveling temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Z{first_layer_height + z_offset} ; Lower to print first layer
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G92 E0.0
G1 E7.5 F700
G1 X60.0 E9.0 F1000.0 ; intro line
G92 E0.0
G1 E-0.5 F900; Retract a little. 900mm/min

End G-Code:

G4 ; wait
G92 E0 ; set extruder value back to 0
G1 E-1 F700 ; Retract 700mm/min
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+5, max_print_height)}{endif} ; Move print head up
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
G1 E-0.5 F700 ; Retract 700mm/min
G4 P300
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+50, max_print_height)}{endif} ; Move print head up
G1 E-2 F700 ; Retract 700mm/min
G4 P300
G1 X0 Y200 F3000 ; home X axis
M900 K0 ; reset LA
G4 P15000
M84 ; disable motors

**Using 250° C**

Start G-Code:

M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.9.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83  ; extruder relative mode
M104 S160 ; set extruder leveling temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S160 ; wait for extruder leveling temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Z{first_layer_height + z_offset} ; Lower to print first layer
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G92 E0.0
G1 E15 F700
G1 X60.0 E9.0 F1000.0 ; intro line
G92 E0.0
G1 E-0.5 F900; Retract a little. 900mm/min

End G-Code:

G4 ; wait
G92 E0 ; set extruder value back to 0
G1 E-4 F700 ; Retract 700mm/min
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+5, max_print_height)}{endif} ; Move print head up
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
G1 E-0.5 F700 ; Retract 700mm/min
G4 P150
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+50, max_print_height)}{endif} ; Move print head up
G1 E-2 F700 ; Retract 700mm/min
G4 P150
G1 X0 Y200 F3000 ; home X axis
M900 K0 ; reset LA
G4 P15000
M84 ; disable motors

Hope it helps!

bubnikv, 2020-10-17:
Settings are better discussed in https://github.com/prusa3d/PrusaSlicer-settings/issues
I bet a duplicate of this discussion is to be found there.
Closing.

closed 2020-10-17 by bubnikv

5axes, 2021-01-16:
tested also with rect_size_x  and rect_size_y without any success

arittner, 2023-10-06:
> tested also with rect_size_x and rect_size_y without any success

In Orca slicer I found this solution:

`G1 X5 Y{print_bed_max[1]*0.8} F{travel_speed*60} ; present print`

So, `print_bed_max[1]` is similar to rect_size (y) - the depth, and `print_bed_max[0]` to x - the width.