prusa3d/PrusaSlicer #1127: Missing header EXTERN.h

stelgenhof, 2018-08-07:
### Version
1.41.0-beta
`git rev-parse HEAD`: 288cd58

### Operating system type + version
Arch Linux 4.17.11

### Behavior
Compiling Slic3rPE from the latest sources fail with the error that the EXTERN.h file cannot be found:

`../xs/src/xsinit.h:44:10: fatal error: EXTERN.h: No such file or directory.`

This actually happened after a system update of Perl was done (2018/08/07).
`perl -v`: perl 5, version 28, subversion 0.

From a quick search on the internet, this missing header file issue has occurred in the past before. I guess this is not so much an issue with Slic3r (as it was compiling fine before the Perl update), rather than an issue with the Perl update. 

Just want to make people aware of this issue in case they are also going to upgrade Perl on their OS.

stelgenhof, 2018-08-07:
Further investigation showed that some manually installed Perl modules had to be rebuild/re-installed:

`WARNING: '/usr/lib/perl5/5.26' contains data from at least 5 packages which will NOT be used by the installed perl interpreter.`

- perl-alien-wxwidgets
- perl-extutils-cppguess
- perl-extutils-typemap
- perl-extutils-typemaps-default
- perl-extutils-xspp
- perl-opengl

After doing so, the issue still remains unfortunately. Even rebooted my system.

mloidl, 2018-08-07:
I had the same issue.
Removing CMakeCache.txt and generating calling cmake again solved the issue for me.

mloidl, 2018-08-07:
Would do a `make clean` prior the new build to make sure there are now old objects lying around.

Had the same issue somewhere in the past but definitely not yesterday.

stelgenhof, 2018-08-07:
@mloidl Already did a complete clean but unfortunately didn't solve the issue. Issue is now the the Perl runtime.

bubnikv, 2018-08-07:
> ListUtil.c: loadable library and perl libraries are mismatched (got
handshake key 0xde00080, needed 0xce00080).

https://stackoverflow.com/questions/45000585/listutil-c-loadable-library-and-perl-binaries-are-mismatched-got-handshake-key
https://github.com/hideo55/Module-Build-XSUtil/issues/12

Good luck guys

I think you need to reinstall the following cpan module.
https://metacpan.org/pod/List::Util

stelgenhof, 2018-08-07:
@bubnikv Thanks! I will check those links out later. Tired after hours spending figuring this one out.

I will close this issue as there is nothing wrong with Slic3rPE itself, but more my environment after I did a system upgrade which included an upgrade to Perl 5.28 (from 5.26). Never experienced this before, actually.

closed 2018-08-07 by stelgenhof

stelgenhof, 2018-08-07:
@bubnikv Just quickly I want to add that I've tried already many of the suggestions on StackOverflow, etc. All the modules are actually installed, it's just that the Perl interpreter tries to pick the wrong/previous one.

vojtechkral, 2018-08-07:
I had a similar problem yesterday, had to recompile a bunch of perl modules _and_ wxwidgets...

stelgenhof, 2018-08-07:
@vojtechkral I did the same, but still encountering issues. I'll keep this ticket up to date if I have more info.

vojtechkral, 2018-08-07:
@stelgenhof You're using Arch too, right? FWIW, this is what I did:

1. Remove Slic3r build directory, remove the `local-lib` directory
1. Install as many perl dependencies using `pacman` and from AUR as possible; the ones that come from AUR apparently needed rebuilding
1. Re-build `wxgtk` from `asp`/`trunk` because of the ABI mismatch
1. Re-build Slic3r

stelgenhof, 2018-08-07:
@vojtechkral Yes, using Arch. I will try out these steps. Thanks!

stelgenhof, 2018-08-07:
@vojtechkral Woohoo! Got it working again. I followed your steps, except number 3. Not sure why, but without that it compiled with no issues and ran without a problem.

As always, big thanks @bubnikv and @vojtechkral !