prusa3d/PrusaSlicer #1129: Need help tuning Single Nozzle MM related Printer and Filament specific settings to fix filament jams during tool change

jimmy00784, 2018-08-07:
### Version
1.41.0 beta linux 64

### Operating system type + version
Arch Linux 64 bit

### Behavior
During tool change, the filament deforms unevenly and results in jams after a few layers of successful prints. Need assistance troubleshooting and tuning single nozzle MM related settings that are present in Slic3r PE.
I've attached pictures of filament ends that I've cut from past different prints as well as my 3D printer for your reference. I can provide more pictures if necessary.
I've also attached my config bundle if that'll help.
Since this feature is so new and unique to Slic3r PE, there is no documentation or resources out there to help troubleshoot the issue.
TIA 

[image]

[image]

#### STL/Config (.ZIP) where problem occurs

[Slic3r_config_bundle.ini.txt](https://github.com/prusa3d/Slic3r/files/2267242/Slic3r_config_bundle.ini.txt)

bubnikv, 2018-09-01:
Sorry, we cannot invest time into tuning your machine. You are on your own.

closed 2018-09-01 by bubnikv

gruvin, 2020-06-23:
FWIW and IMO ... @jimmy00784 put considerable effort into making his question VERY clear. Thank you. He certainly was not asking for what to do to tune his printer. He made this painstakingly obvious and must have been quite offended by the brush off. I hope @bubnikv was just having a bad day. 🤷 Now, with that out of the way ...

As far as I can see [two years later] this was and still **is a valid issue**, worthy of developer attention.

Since there is apparently still no documentation and nothing I can find in the [now] PrusaSlicer UI (which is fine -- devs can't do everything, especially for free) and in order to help out with that, we first need to know: 

- Can these settings even be changed by the user -- in the UI or anywhere else? It seems not. 
- If not, then perhaps this issue should be upgraded to at least a feature request?

We could help write the documentation, if we knew where to look for related source code variables or .ini parameters.

jimmy00784, 2020-06-24:
Thanks @gruvin
Lack of assistance and limited documentation in this area prompted me to continue my own research into this matter.

The issue was being caused by a number of factors:
1. Heat creep issue
2. Filament not being pushed or pulled with enough pressure through the PTFE tube
3. Slack in the PTFE tube 

I made the following modifications to this printer, after which, I no longer encountered the issue.
1. Changed PTFE tubes to high-quality tubes from Capricorn.
2. Changed all four extruders to Titan clones from Th3D Studio (3-to-1 gear ratio).
3. Changed the nozzle from 0.4 mm to 0.5 mm.
4. Changed the heat break to high-temperature all-metal PTFE heat break from Th3D Studio.

I built a hypercube printer with the same MMU setup and these modifications proved useful there as well.

gruvin, 2020-06-24:
@jimmy00784 That's great news. 

I actually came back to delete/edit my comment, as  finally found where all the settings were, in Prusa Slicer v2.20 anyway. THen noticed you had replied already. 

`Filament` / `Advanced`, under `Wipe tower parameters`

I am yet to quite figure out how to massage those settings to match my tuned G-code version, but it's great that they're there.

Ironically, mine is a Hypercube derivative as well. :-) I'll check out your profile here to see if you have a blog or something.