prusa3d/PrusaSlicer #1131: Gyroid infill with, it´s impossible to change ...

ChiliApple, 2018-08-08:
### Version
1.41.0 Beta+

### Operating system type + version
Mac OS 10.13.6

### Behavior
* since 2-4 Versions it´s impossible to change the infill with on gyroid, works only on the other infill´s

[image]
[image]

thx, best Udo

supermerill, 2018-08-09:
How do you "change the width" ? with the parameter in advanced->infill?

bubnikv, 2018-08-09:
The issue here is, the gyroid infill as of now is printed with a bridging
flow, so it does not respect the infill extrusion line width.

KD7LRJ, 2018-08-09:
The "Combine infill every:" also seems to cause problems with gyroid infill.

bubnikv, 2018-09-01:
@supermerill 
> Yeah, it's because it's using "bridge" flow in slic3rPE.

Wasn't this your initial implementation? Does the non-bridging flow produce sufficient overlaps?

supermerill, 2018-09-03:
> Wasn't this your initial implementation?

Yes, it was in my initial implementation, as I used 3DHoneycomb as a baseline.
Month later, i removed it because it create more problems than it resolve. 

> Does the non-bridging flow produce sufficient overlaps?

Yes. (result below are with 0.9 bridging reduction & 0.4 line width)
At 0.35mm, it increase it from 0.13 to 0.15.  (not-bridge extrude more than bridged)
At 0.2 mm, it reduce the section from 0.13 to 0.09  (I printed almost only 0.2 and no problem with that).
At 0.05mm, the bridge flow completely destroy the infill (because there are no space for it).

note: I wanted to use the dont_connect to let user say "don't connect the lines, i want to create an art piece" but I have to create a new gui button for that. I think i will clean the fill api & create some more gui option for it in the future.

bubnikv, 2018-10-23:
The bridging flow was disabled for Gyroid infill with b429b39abedf13ad77839aa9cc478e28225c1a2c
The fix will be in 1.41.1 final.
Closing.

closed 2018-10-23 by bubnikv