prusa3d/PrusaSlicer #1133: infill separately for TOP / BOTTOM Layer

ChiliApple, 2018-08-08:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* last request :)
Please make it possible for us to be able to select the infill separately for TOP / BOTTOM Layer

_Is this a new feature request?_ YES

THX best Udo

supermerill, 2018-08-09:
Already done in my fork https://github.com/supermerill/Slic3r
the commit: https://github.com/prusa3d/Slic3r/pull/821/commits/f9344a00e37e179e44f970edd65fd124a9370a96

duplicate of https://github.com/prusa3d/Slic3r/issues/475

ChiliApple, 2018-08-09:
looks good, but no ready use app for Mac OS, i must be compile your fork?

supermerill, 2018-08-09:
Yes

bubnikv, 2019-02-22:
Implemented with ab30370fb4bfa92d8ae77419686e2b6cde923fcf
Thanks @supermerill for the initial implementation.
Closing.

closed 2019-02-22 by bubnikv