prusa3d/PrusaSlicer #1137: Cannot enforce support (Generate support unselected)

buttc, 2018-08-10:
### Version
Version 1.39.1-prusa3d-win64

Windows 10 Home OS build 17134.165

Under Print Settings> Support material>
Generate support material - UNTICKED.

Enforce support for 30 layers
DOES NOT NOT CREATE SUPPORT!!

[config.zip](https://github.com/prusa3d/Slic3r/files/2279285/config.zip)

[Arch long legs with top pins.zip](https://github.com/prusa3d/Slic3r/files/2279280/Arch.long.legs.with.top.pins.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

buttc, 2018-08-11:
Just to add the doorway is printed vertically, the pins are the only part in contact with the bed.

jkavalik, 2018-09-01:
"Generate support material - UNTICKED" afaik means that support structures generator is entirely disabled so enforcing does nothing - you have to enable supports (but you can use the threshold of 1 to get as little of them as possible, 0 means automagic so not usable here) and then enforce what you need:
[image]

bubnikv, 2019-01-08:
Duplicate of #470, closing.

Currently you have to have the supports enabled for this feature to work. I am closing this issue.

closed 2019-01-08 by bubnikv