prusa3d/PrusaSlicer #1139: Fan turns off during first two layers of non-build plate supports

tjcolejr, 2018-08-12:
1.41.0-beta+win64

I have no idea how long this has been happening,

In my filament settings, I have the fan set to turn off during the first 2 layers of the model and this is functioning correctly.

I am printing supports that begin at layer 7, starting on a top surface of the model, and I noticed the fan was turning off when the first 2 (possibly 3) layers of the supports were printing.  Below is a snip of the gcode showing the M107.

I think this is an error and needs to be either a selectable feature, or corrected because it bonds the bottom of the supports too well to the top of the model.

G1 X154.541 Y73.695 E0.02222
G1 X154.558 Y73.707 E0.00031
G1 X155.076 Y74.076 E0.00954
M107
;BEFORE_LAYER_CHANGE
;1.56

G1 F8640
G1 X154.558 Y73.707 E-0.14664
G1 E-0.45336 F2100.00000
G1 Z1.560 F10800.000
;AFTER_LAYER_CHANGE
;1.56
; PURGING FINISHED
G1 X149.572 Y73.731
G1 E0.60000 F2100.00000
G1 F1800

tjcolejr, 2018-08-12:
Additionally, this DOES NOT happen if there are ALSO supports directly on the build plate.

bubnikv, 2018-09-01:
@FidelCapo would you please test this?

FidelCapo, 2018-09-03:
@bubnikv I can reproduce this issue

tjcolejr, 2018-09-03:
Here is the relevant portion of my startup gcode:

M104 S130 ; set extruder temp
M140 S60; set bed temp
G28 W; home without mesh bed leveling
G1 X125 Y100 F10000; move to the middle of the bed
M190 S60; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
G80; mesh bed leveling
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

And FYI, it completes the startup faster than the standard gcode.  So if
waiting is a problem.....
Notice I set the extruder to 130 first.  I have timed this so the mesh bed
leveling completes before oozing begins, and the heating of the bed never
stops so everything happens concurrently with the longest operation.

closed 2026-09-29 by prusaslicer-bot[bot]