prusa3d/PrusaSlicer #1140: No Preset for Prusa Easy-ABS / Filament-PM ABS-T

stahlfabrik, 2018-08-13:
This is a feature request (and a little complaint:-) aka cry for help)

I want to use my Prusa Easy ABS spool that I have. Problem: There is a "Prusa ABS" Profile, but no "Prusa Easy ABS" profile in Slic3r.

When I print using the Profile "Prusa ABS" the prints come out unusable bad: completely deformed and molten.

The "Prusa ABS" profile uses 255C on hotend. Easy ABS is specified for 230 to 250C. But just lowering the temperatures does not lead to success.

I know that slowing down and using the nozzle fan improves quality - but the specification of ABS-T says "NO" to the use of a part cooling fan - so I fear I might get weak layer adhesion.

What I would wish is Profile for Easy ABS that works just out of the box. But of course, if I need to turn speeds etc. I would be happy to just get of settings that have proven to work for Prusa Research. I mean - you know your filament better than anybody else:-)

Thank you! Any advise greatly appreciated.

jindrichbenes, 2018-09-01:
Prusa abs = prusa easy abs
Suggested dimensions when printing with abs are 3 to 8 cm. Are you by any chance printing something small?

stahlfabrik, 2018-09-02:
Yes and no:-)

I learned a lot about printing in ABS-T. I agree that the profile in Slic3r does work. I learned how to add a 1perimeter cylinder and specifically print that at 5mm/s. This way by varying the diameter, I was able to add cooling time. So I had cylinders for 10, 20, 30 and 40s. With that trick, I managed to get stuff, like the fan nozzle (for MK3) printed. I also tried to print the extruder parts with that method in mind. But I failed here: The overhangs on the window and where the filament enters the gears looked very bad. I also tried to print the X axis parts. Depending on the bed placement some parts got deformed because of uneven cooling.
The thing is: if I add 20 to 30% cooling fan, the prints come out perfect optically - but I guess the strength is bad then. So I do not know: ABS-T is not for me, when it comes to print replacement/upgrades to my only 3d printer. In order to achieve dimensional accuracy I must user PETG...

I close this now, as I learned the profile is fine, but still ABS-T beats me:-(

closed 2018-09-02 by stahlfabrik