prusa3d/PrusaSlicer #1141: [Feature Request] Insert pause and filament change at specific layer height

subjektivdk, 2018-08-14:
**M600**
Pause for filament change

**M601**
Long pause

Something like in prusacontrol

hexane360, 2018-08-18:
This can be done using conditional g-code. Try something like the following in your "Layer change g-code":
```
{if layer_z >= 15 && layer_z < 15 + layer_height}M600 ;pause{endif}
```
For more info, the macro language is documented [here](https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language).

bubnikv, 2018-09-01:
@YuSanka is working on a UI similar to the Prusa Control. It will most likely be part of 1.42

tbrummel, 2018-09-04:
Since this is being worked on, I figured I'd ask here if the bug relating to color change extrusion will be fixed as well.  When an M600 is encountered, the printer stops, changes filament, and then asks if the filament has been changed successfully.  When you say "Yes", then the printer extrudes more filament just before going back to the object to print!  This, obviously, is bad news - you have to be quick cleaning the tip of extra filament or it will be dragged into your print and ruin it.  

This bug has been there for a long time -- and I suspect it's a firmware thing.  I looked at the code once and it did seem to be extruding an amount to 'prime' things -- but that step is unnecessary at a color change.

Quintox303, 2018-09-14:
@tbrummel This is the intended behaviour so it's not really a bug, although it may be undesired for some. For me I just make sure I'm ready with pliers to grab the primed filament before it moves back over the print. I do agree that there must be a more elegant solution. 

Either way, that would be a firmware modification and not something related to the implementation in slic3r

bubnikv, 2018-12-11:
The M600 aka "color print" was implemented in Slic3r PE 1.42 alpha to be released soon. Closing.

closed 2018-12-11 by bubnikv

alyyousuf7, 2019-03-13:
@bubnikv, only the M600 was added in Slic3r PE 1.42 alpha.

It would be great to have M601 - Long pause for inserting magnets or whatever. [This video](https://www.youtube.com/watch?v=pBPxY1HdPG4) shows a use case.
Adding a pause from Slic3r PE in the same way as color change would be great.

We can have a "pause" button for M601 beside the "+" button for M600.

ggrote, 2019-04-01:
Would be even better to have something like this for inserting magnets:
```
G1 X10.000 Y200.000 E0; parking position
M1; user stop
M105; return to current temp
```

TheNetStriker, 2019-05-22:
It is now possible to add a color change in PrusaSlicer 2.0, but it would be great if also a simple pause could be added like in the example above. Are there any plans to implement this?

bubnikv, 2019-05-22:
https://github.com/prusa3d/PrusaSlicer/issues/1899