prusa3d/PrusaSlicer #1142: Send to printer button disappears in 1.41.0-beta

geeksville, 2018-08-14:
### Version
1.41.0-beta+linux64

### Operating system type + version
linux 64 bit, ubuntu 16.10

### Behavior
The "send to printer" (octoprint) disappears.  Steps to repo is quite easy:

1. launch slic3r - see button, if you load a model you can click to send to printer.
2. however, if you use the filament dropdown button to change the filament type, the button immedately disappears.  
3. Only way to get it to come back is quit and relaunch slicr3

This problem did not occur in the 1.40 release.

geeksville, 2018-08-14:
Oh - I found the problem I was using a filament.ini file from a friend, which redefined some non filament settings.

bubnikv, 2018-08-15:
> Oh - I found the problem I was using a filament.ini file from a friend, which redefined some non filament settings.

Would you be more specific, please? Is there some specific scenario or legacy configuration data that we should be aware of?

Would you please provide your filament.ini? We should avoid such issues.

geeksville, 2018-08-15:
Sure no problem.  Steps to repro:

1. Receive a config bundle from someone else (which contained printer, filament and print ini entries)
2. Not realize that the file I was provided was not just a filament.ini
3. Install that file in my ~/.SlicerPE/filaments directory
4. Use the file for printing (along with my other printer and print settings)
5. The bug is that the GUI lets you change print/printer settings, but they are actually not having any effect, because the (extra - because this was a 'config bundle') entries in the filament.ini file are silently overriding all the other settings.

Recommended behavior: validate printer/print/filament files and show the user a message like 'inappropriate configuration settings were found in bobsfilament.ini - ignoring them (did you instead mean to import this as a config bundle?)'.

bubnikv, 2018-09-01:
@FidelCapo would you please verify this issue?

FidelCapo, 2018-09-03:
I can confirm that if the config bundle is selected as a filament preset, it overwrite some print or printer settings.

bubnikv, 2018-09-27:
fixed with 
b338eb0

closed 2018-09-27 by bubnikv