prusa3d/PrusaSlicer #1143: Modifier shapes create extra perimeters inside the object

fpikus, 2018-08-16:
### Version
1.41.0-beta-win64

### Operating system type + version
Windows 10

### Behavior
When a modifier shape intersects the object vertically (part of the vertical wall of the modifier is inside the object), extra perimeters are created at the boundary of the modifier. Horizontal walls are no problem even if they are inside the object.

#### STL/Config (.ZIP) where problem occurs
[Test.zip](https://github.com/prusa3d/Slic3r/files/2292291/Test.zip)

DeepSOIC, 2018-08-29:
+1

Examples where it is useful...

I am making a handle for screwdriver bits. Or for a tap. or for a potentiometer...
The hole should be strong, so I'd like some extra perimeters. I add a cylinder, and apply 5 perimeters modification...
[image]

And I get this. 7 perimeters somewhere in the middle of the part, wasting plastic and serving no purpose.
[image]

one more example to follow...

DeepSOIC, 2018-08-29:
And another example. I want to make a speaker enclosure. 3D-printing has a fantastic feature, sparse rectilinear infill. Being air permeable, it makes a good filler for the acoustic chamber. It dampens standing waves inside, as well as makes walls extremely rigid, allowing me to make walls thinner and make the speaker lighter. 

But, it's hard to achieve an opening for the driver. I would like to remove some perimeters on the wall, but modifier shape does not quite achieve that:
[image]
It is SOOO CLOSE, but not right.  I usually have to model the volume for infill in cad, and then slice the shell and the volume as separate solids.. but it just takes too much time to model (my enclosures are usually not as simple as this one)… and it's hard to get perfect number of perimeters to fit into the thickness of modeled shell. It is complicated to the point that I prefer just to print the enclosure with walls everywhere, and then rip out the unwanted walls with Dremel or something... 

Would be great if I could pull this off right in Slic3r, just with a bunch of modifiers..

bubnikv, 2019-11-01:
Splitting of perimeters and infill at the top / bottom surfaces was fixed in PrusaSlicer 2.1.0. Closing.

closed 2019-11-01 by bubnikv

schaudi, 2020-03-02:
@bubnikv no the issue is not fixed - u don't get the issue descript here, which has nothing to do with that. For example, if I tell PrusaSlice to print that tail section of this pikachu (tail on the right), it adds perimeter in between. But this doesn't help with stability. 
[image]

 The same in the latest stable, all current alphas and beta.

bubnikv, 2020-03-02:
@schaudi Please provide a 3mf.

schaudi, 2020-03-02:
@bubnikv Of course, here it is. The Problem is even visible in Slic3r Manual https://manual.slic3r.org/advanced/modifier-mesh as shown in the image
[image]
[Pikachu.3mf 2.zip](https://github.com/prusa3d/PrusaSlicer/files/4275095/Pikachu.3mf.2.zip)

bubnikv, 2020-03-03:
@schaudi You are right, that changing perimeter speed leads to such artifacts. Currently neither PrusaSlicer nor the upstram Slic3r allow one to change parameters of just a piece of perimeter. The change is applied to a region, around which a perimeter is drawn. The only exception to this rule currently is a situation, where the override parameters modify the infill only. In that case PrusaSlicer is smart enough to generate single perimeter around the merged infill regions.

Steven1994, 2022-01-26:
Any fixes for this problem yet? I need to add a modifier in the middle of my part for my purpose, but then it generates random perimeters there...

pjsmith72, 2022-08-17:
I need this fix too!
I've wanted to use modifiers to add extra perimeters, but I hate to literally slice my part in half to achieve this!

It all has to do with conditions where the Modifier part's outer shell cuts through the "Part" part. I think there should be special rules to keep the part (as a whole) intact (or the outer perimeter(s) of the part, at the very least).

I can think of a few solutions. Let me know what you guys would prefer.
Or maybe there could be a drop-down option set (or two) for any/all of these solutions.
(See image below)
The perimeters settings are:
default: 3 perimeters
modifier: 10 perimeters

Solution A - Black lines)
I think the external perimeters should remain ongoing between both the modifier part and the "Part" part.
This is my biggest "gut feel" requirement!
Solution A - Blue lines)
These would be "required" to maintain the 10 perimeters in the Modifier part, around the entire Modifier part.
Solution A - White lines)
These would be "required" to maintain the 10 perimeters in the Modifier part, around the entire Modifier part where the modifier part's outer shell cuts through the "Part" part.
It would be nice if we could have a drop-down to enable or disable these white lines. I know it would turn the "10" perimeters into 7, but I'm okay with that (especially if I had the option).
Solution A - Purple lines)
These would be "required" to maintain the 3 perimeters in the non-Modifier "Part" part, around the entire non-Modifier "Part" part where the Modifier part's outer shell cuts through the "Part" part.
It would be nice if we could have a drop-down to enable or disable these purple lines. I know it would turn the "3" perimeters into 0, but I'm okay with that (especially if I had the option).
Not to mention: Before the Modifier part, there were no internal perimeters here in the first place, so I'm very okay with getting rid of these!

Solution B
(only change is "Solution A - Blue lines" and "Solution A - White lines")
I've been using double quotes for "required", because none of this is actually required. I think we're all asking for a change.
It would be nice if we could have a drop-down to switch between Solution A, Solution B, and what it is doing now!

Solution B - Black lines)
Same as Solution A
I think the external perimeters should remain ongoing between both the modifier part and the "Part" part.
This is my biggest "gut feel" requirement!
Solution B - Blue lines)
This is the main difference between Solution A and Solution B.
I'm fine with them just terminating.
It would be nice if we could have a drop-down to switch between Solution A, Solution B, and what it is doing now!
Solution B - White lines)
I don't even think these are required. If we went with Solution B, I'd propose the default being no white lines.
It would be nice if we could have a drop-down to enable or disable these.
Solution B - Purple lines)
Same as Solution A. I don't think these are necessary. But if possible, it would be nice if we could have a drop-down to enable or disable these.

[image]

[image]

Reference: Types of "parts":
[image]

[image]

stepikovo, 2022-08-24:
We are aware of the not ideal current situation and we might rework it one day.

ffpp2003, 2024-02-17:
This is still an issue on PrusaSlicer 2.7.1. Has any progress been made on the topic?
[image]
In this model I want the inner perimeters to be 5 perimeters thick to improve strength, because a bolt and a nut are going to be inside and will experience quite some force. The issue also appears on the top and bottom layers:
[image]

[Cuerpo2.zip](https://github.com/prusa3d/PrusaSlicer/files/14319470/Cuerpo2.zip)

Joerideman, 2025-03-28:
I think the issue is not closed:

When painting my object it also adds additional unwanted perimeters. Making the print use more plastic than expected.

[image]

[image]