prusa3d/PrusaSlicer #1145: Dark Mode on macOS Mojave and Windows 10 and keyboard shortcuts for Mac

Patrik-svobodik, 2018-08-17:
Hi, would it be possible to add dark mode and shortcuts for Mac?

bubnikv, 2018-09-01:
@YuSanka @FidelCapo would you guys evaluate how our current Slic3r PE looks in dark mode on OSX Mojave? Thanks.

@Patrik-svobodik What shortcuts? It is better to not mix two issues in on github issue, better to create a new one.

@Patrik-svobodik Do you have a hint, what is not working in the dark mode?

Patrik-svobodik, 2018-09-01:
@bubnikv Everything looks OK in dark mode, but it would look much better if the app UI turned dark as well, I meant keyboard shortcuts

bubnikv, 2018-09-01:
I don't get it. You want the keyboard shortcuts to turn dark? Would you please mark it on your screenshot?

Patrik-svobodik, 2018-09-01:
I want dark UI and keyboard shortcuts to be added

bubnikv, 2018-09-01:
Nerozumím, co chcete. Jak spolu souvisí dark mode a keyboard shortcuts?

Patrik-svobodik, 2018-09-01:
vůbec spolu nesouvisí, jen že bych chtěl přidat klávesové zkratky a taky Dark mode pro macOS Mojave, neměl jsem to psát do jednoho issue

bubnikv, 2018-09-01:
A jaké klávesové zkratky?

Patrik-svobodik, 2018-09-01:
Třeba backspace pro smazání objektu, alt/cmd + backspace pro smazání všech objektů

Patrik-svobodik, 2018-09-01:
Mimochodem, place on face se mi nezobrazuje v nejnovější betě slic3r PE

FidelCapo, 2018-09-03:
@Patrik-svobodik nezaznamel jsem nezobrazování "place on bed" ve verzích 1.41.0-beta2 ani 1.41.0. Jste si jistý, že jste nebyl v "Preview" okně? V náhledu se tato funkce nezobrazuje.

Patrik-svobodik, 2018-09-03:
měl jsem nejnovější verzi z gitu, pak jsi tam nahrál novější a tam to je

bubnikv, 2018-12-12:
We seem to get the dark mode on OSX only if we compile Slic3r against a newer OSX SDK, but then the dark mode is inconsistent.

It looks like the wxWidgets do not yet support the OSX dark mode correctly.
https://trac.wxwidgets.org/ticket/18146

I think it is sensible to wait until the wxWidgets support of the OSX dark mode is polished.

vojtechkral, 2019-01-10:
It seems recently wxWidgets polished Mojave support somewhat and dark mode seems to work as far as I can tell. However, to have the support, you need to build Slic3r yourself (see also #1421). At the moment we don't provide Mojave builds as our infrastructure is not configured to do that and even if it were, it would not be straightforward to support both Mojave features as well as older Macs in one bundle.

bubnikv, 2021-04-07:
implemented

closed 2021-04-07 by bubnikv