prusa3d/PrusaSlicer #1149: Add ability to set 'parameters' in the custom gCode section

rchancey, 2018-08-18:
### Version
_Version of Slic3r used goes here_
1.41.0 beta

### Operating system type + version
Mac 10.13.6 high Sierra 

### Behavior
I tried to set the bridge_fan_speed and bridge_speed so I can change those between layers. I am using a print tower and trying to change not just some of the normal temp (M104) or retraction(M207) but also want to override some of the 'parameters' 

I assume I can just use the parameter names and set the values .. such as
bridge_speed=50

but that does not seem to have any effect.

this may be a new feature if it is not already supported

this would allow us to use temp towers and to tune in the settings  MUCH faster. 
I am using this: https://www.thingiverse.com/thing:2729076

and code like this:
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
{if layer_z==1.6}
; T tower floor 1
M207 S1
{elsif layer_z==11.6}
; T tower floor 2
M207 S1.5

etc.. 

can see it would be nice to set a variety of settings. .. one at a time of course.. to really dial in a filament. 

I may be using the wrong syntax... please provide an example of setting parameters 

.. especially things like perimeter_speed, bridge_speed, bridge_fan_speed.. some of the normal ones we tend to tweak.

bubnikv, 2018-09-01:
One cannot set variables by the conditional G-code as of now, and it would be quite tricky for most of the parameters. At the time when the conditional G-code is being interpreted, most of the slicing has already been done, so the change of most parameters would produce no effect. There is a handful of parameters for which it may make sense, but then introducing such a feature would only increase the amount of questions on us poor developers, why this or that does not work.

rtyr, 2021-01-21:
Seems already answered. I would say it is better to use height range modifiers or custom g-code at specific layer (can be done using UI).
Closing.

closed 2021-01-21 by rtyr