prusa3d/PrusaSlicer #1151: Outside print volume error, some are detected as to large.

gjelsoe, 2018-08-19:
### Version
1.41.0-Beta+win64

### Operating system type + version
Windows 7

### Behavior
Some large objects are detected as to large.
At the screenshot following things are done.

1. Add object
2. Rotate X 90 deg
3. Rotate Z 20 deg
4. Manual arrange or Arrange button

[image]

Works in version 1.40.0+win64

#### STL/Config (.ZIP) where problem occurs
[files.zip](https://github.com/prusa3d/Slic3r/files/2300316/files.zip)

dikodahan, 2018-08-19:
I am also running the 1.41.0 beta version on the latest MacOS (10.13.6) and am experiancing the same issue.

bubnikv, 2018-09-01:
This should be fixed in 1.41.0 final release.

closed 2018-09-01 by bubnikv

sersmax, 2018-09-14:
I have this issue in Version 1.41.0 now...
It is in MMU mode, but i assume that this is not the reason for this error.
[image]

FidelCapo, 2018-09-17:
@sersmax Can you send me this MM object please? Thank you

mrardon, 2018-09-21:
@FidelCapo I am having this issue as well in 1.41.0+ on Mac. But I have also noticed something odd.

- Parts from: https://www.thingiverse.com/thing:3078915
- Load object: cs_long_bottom.stl
- Select object and goto Settings
- Add part: cs_long_top.stl

Get "Detected object outside print volume". If using with the 1 long piece part is is fine. If I use the short version (cs_bottom.stl and adding cs_top.stl it works)

The odd thing I found was when looking at the "face-to-surface" option and seeing what faces it thought there were. They seem to be removed from the object itself and those faces appear to be outside the print area. I have attached the images of the faces showing the issue.

[image]

[image]

sersmax, 2018-09-24:
> @sersmax Can you send me this MM object please? Thank you

Yes of course:
[Triangle.zip](https://github.com/prusa3d/Slic3r/files/2411225/Triangle.zip)
maybe i should mention that this is on a bigger Printbed with the size of 290mmx300mm.
But i can imagine that this does not depend on the size of the bed, You can make the object smaller that it fits the planed bad but will still give the said error.

Sorry for the late reply. The problem got even more weird:
If i add ONE file first and double click it to add the second file i get this problem. 
But if i add both files in one add operation it does work just fine.

FidelCapo, 2018-09-25:
@sersmax Thank you. Unfortunately, we know about this error. I will use your object as a test for fix.