prusa3d/PrusaSlicer #1155: Will you ever fix the reversed perspective?

dlevine999, 2018-08-20:
### Version
1.40.1+win64
_What OS are you using, and state any version #s_
win 10 64
### Behavior
The perspective of the 3Dview is backwards.  e.g. the front of the bed is smaller than the back.
_Is this a new feature request?_ 
Sort of. I know Slic3r has always done this but it seems much worse and more disorienting in the latest version.

Panayiotis-git, 2018-08-20:
It seems strange. But it is in orthographic view which is not wrong. I think you suggest to also add an option for perspective view.

dlevine999, 2018-08-20:
It is not an orthographic view. With a true orthographic view parallel lines (like the sides of the build plate) would  remain parallel. I don't know what to call the view you actually have; I have never seen it used anywhere else. I strongly suspect that the original programmer was trying to do perspective and got it backwards.
 -- David R. Levine = dlevine999@yahoo.com

vojtechkral, 2018-08-22:
> With a true orthographic view parallel lines (like the sides of the build plate) would remain parallel.

But they are parallel. My theory is it looks weird because the brain expects perspective and adjusts for that.

dlevine999, 2018-08-23:
Well, I'be ******. It is an optical illusion. One I knew about as a kid - many decades ago - and had forgotten.

It's still very annoying though. so this is a new feature request.  Please add perspective view.

lixxbox, 2018-08-24:
I would also prefer a perspective view.

bubnikv, 2018-09-27:
This is something for @enricoturri1966  to look into.

henrikkoch, 2019-03-10:
Please add perspective view as an user option to use or not

bubnikv, 2019-05-14:
We have a good news, @enricoturri1966 started working on perspective camera.

krhackbarth, 2019-05-18:
is there a separate site for suggesting and voting up/down enhancements?  GitHub just doesn't seem to be the right place for that.

Serveurperso, 2019-06-16:
Please !!! Isometric view drives me crazy !!!!! Please enable perspective by default ! We don't need a GUI configuration option because isometric is totally useless for a non-CAD preview.

lukasmatena, 2019-06-17:
1. The work on the perspective camera has started already, as mentioned above. There is no need to upvote it anymore.

2. Maybe you don't need a configuration option. But you would be surprised what other people sometimes consider as "useless" or "must have".

bubnikv, 2019-08-01:
Perspective camera will be implemented in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv

MucaroBoricua, 2019-12-14:
This issue is still happening in PrusaSlicer 2.1.1

lukasmatena, 2019-12-14:
What issue? Perspective camera was added into 2.1.0. Just enable it in preferences.

MucaroBoricua, 2019-12-15:
You are right.  I spoke too soon.
I had just upgraded from Slic3r 1.3 something.  (I had my 3D printer in storage for 3 years).
Then I noticed that the bed looked weird so I looked it up in the issues and found this.
Hours later I found the K command to change from ISO to Perspective views.
And it all makes sense now.