prusa3d/PrusaSlicer #1156: Brim destroyed by support aka. Enable supports on top of brim

sav0nar0la, 2018-08-20:
### Version
1.41.0-beta+linux64

### Operating system type + version
Mint 18.3

### Behavior

the piece to be printed has little contact area to the bed, therefore brim is needed. 
Because of objects geometry, it also needs supports.

The software gives priority to the supports, so that in the first layer the brims are removed.

As result, warp occurs (lack of brim) and the piece cannot be printed.

I have played with all possible controls, but the supports *always* removes the brim. 

Workaround:
Add a temporary support to the piece so that the brim is replaced, and enable supports "everywhere".

This would be a new feature request:
Allow supports on top of brim

_Is this a new feature request?_
yes
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-09-26:
fix in supermerill/Slic3r#389d3ee4d57d55bc65f2bbeb07ec89ed35618fcc

dmyers7, 2019-03-31:
The point of brim is to connect to the object like when printing ABS.  We can't have support interrupt the brim, leading to warping as the author wrought.  Please elevate to one of the release candidates.

ManIkWeet, 2019-05-15:
I don't see the fix, is it available in the normal settings or do I have to do something fancy/hidden?
My supports can't work with a nonsolid first layer :/

tkircher, 2019-05-23:
Looks like that patch never made it into v2.0. It should though. It's a really essential option, since it's very inconvenient to have to manually add a brim to the STL instead of being able to generate one.

bubnikv, 2019-05-23:
Somebody has to look into the supports in a one long session. Unfortunately we did not have time for it until the PrusaSlicer 2.0.0 release was out. Now we need to implement the Undo / Redo (one feature that people are crying for the most) and then we will have some time on our hands to concentrate on other things like the supports.

ManIkWeet, 2019-05-23:
Fair enough, guess we'll have to wait patiently...

tkircher, 2019-05-23:
Undo is nice too. ;)

eowindel, 2019-08-16:
Yeah it is something that often annoy me. I can do it externally in mershmixer but it would be nice to solve this.
[image]

RCP1, 2020-04-15:
Kind reminder... This issue leads to unavoidable warp for ABS parts, which need support outside their footprint.

quiljw01, 2020-05-11:
Hoping there will be a fix for this.  I'm having the same issue trying to print a wheel.  I need both brim and supports.  But the software only allows one or the other.  Conceptually this is a very simple idea to resolve.  

@sav0nar0la  I don't understand your temporary fix, do you have any screenshots? Have you since found a better solution?  I'm a printing noob so i guess my next option is to see if other software can do it?  

[image]

[image]

sav0nar0la, 2020-05-11:
@quiljw01 : By chance i have seen your email. The idea is simple: add the brim as part of your piece, in the case of the wheel add a thin cylinder to its base (wide enough to get the required adherence to the plate, perhaps two or three layers high). This brim must be removed after printing. Then you have to enable the supports "everywhere" ( also on top of your added brim ) and that should print without problem. Unfortunately I don't have screenshots . . good luck

quiljw01, 2020-05-11:
@sav0nar0la  ahh. yes i understand. good idea i'll do that. thank you!

vtjballeng, 2020-08-08:
This remains a problem 
[image]

bubnikv, 2021-02-23:
IMplemented with 055d232

closed 2021-02-23 by bubnikv