prusa3d/PrusaSlicer #1157: Vase mode configuration not valid

bilobro, 2018-08-23:
### Version
1.40.1

### Operating system type + version
macOS High Sierra 10.13.4

### Behavior
Wanted to print a model is vase mode and i am getting this error.
'Configuration not valid. The spiral vase option can only be used when printing single material objects'
I have tried with a few different stl files and i am getting the same message.  I have also tried to split the model but it says it only contains one part.  This error prevents me from slicing the model

#### STL/Config (.ZIP) where problem occurs

[VaseModeError.zip](https://github.com/prusa3d/Slic3r/files/2313025/VaseModeError.zip)

lukasmatena, 2018-08-23:
I didn't manage to reproduce the issue. As far as I can tell, the model you provided slices correctly in 1.40., 1.40.1. and current master with spiral vase mode enabled. But it also looks as being a part of some bigger model. Please, provide a complete 3MF where the problem occurs.

Generally speaking, the error message that you report means that the object is composed of several parts (each with possibly different config, such as the assigned extruder etc). The Split function cannot help here (it does something else, not to mention you usually don't want to remove any of the parts). What you would need is a "merge" function, which is not part of Slic3r. There is a simple workaround though - Export plate as STL, delete your object and load it again from the STL you created. The inner logical parts are now merged, because there is no way to save them in a simple STL file.

bilobro, 2018-08-23:
Looks like it was just an issue with my computer restart of my comp and slic3r fixed the problem. thanks tho!

closed 2018-08-23 by bilobro