prusa3d/PrusaSlicer #1158: Extra solid layer added across top of part that is supposed to be open

geodave810, 2018-08-23:
### Version
_Version of Slic3r used goes here_1.41.0-beta
_Use `About->About Slic3r` for release versions_1.41.0-beta+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 7 Pro Build 7601: Service Pack 1

### Behavior
* _Describe the problem_When setting variable layer heights in the Layers dialog for this STL file, it adds an extra Bridge infill across the top when the STL file shows it open.  This STL file was generated with openscad & then run thru netFab 7.4 to fix any errors.  I need .1mm height for just the threads, so I wanted to save print time by printing the non-threaded portion at .2mm height.  My workaround is to print the whole object at .1mm height.
* _Steps needed to reproduce the problem_Open the attached STL file & set the variable layer as shown in attached image
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ I expect the top internal threaded portion to be open at the top
* _Actual Results_I get one bridge infill layer across the top
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[IdleArm_LegAxle_2x_Rev1AndConfig.zip](https://github.com/prusa3d/Slic3r/files/2313493/IdleArm_LegAxle_2x_Rev1AndConfig.zip)

bubnikv, 2018-08-23:
Will the extra surface disappear if you scale the objects a bit up? Or if
you change the layer height just a tiny bit?
I believe the problem lies in how Slic3r handles object surfaces exactly
matching a slicing plane. If you change one of the parameters above, the
last slicing plane will no more overlap with the top object surface anymore.

I am working on a fix, but it is quite complex and there is a risk, that
fixing one issue will make some other issues worse.

geodave810, 2018-08-23:
I just changed my height variable of the cylinders in openscad & that works & is an easy work around now that I know what the problem is. Take your time fixing that bug if it is a complex problem.  I really like that feature of being able to adjust the layer heights and have been using it quite a bit the last few months.
Thanks.

bubnikv, 2018-09-01:
I bet it is the same issue as https://github.com/prusa3d/Slic3r/issues/1166
#831
#895
#1102

As noted in https://github.com/prusa3d/Slic3r/issues/1166 :

There is a known issue with the current Slic3r, if a slice cuts exactly at horizontal surfaces of a model. I am working on a fix, but my current prototype works worse on some broken models, so it will take some time.

This is the current branch trying to fix your issue.
https://github.com/prusa3d/slic3r/tree/vb_slicing_fix