prusa3d/PrusaSlicer #1159: Different Layer Heights for multiple Objects

geodave810, 2018-08-23:
### Version
_Version of Slic3r used goes here_1.41.0-beta

_Use `About->About Slic3r` for release versions_1.41.0-beta+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 7 Pro Build 7601: Service Pack 1

### Behavior
* _Describe the problem_If I have 2 objects & set a variable layer height for one object, the other object uses the layer height in the print settings unless I set a variable layer height on the 2nd object to match the 1st object.
* _Steps needed to reproduce the problem_ Add both objects to the slic3r & set the variable layer heights for the tall one & you can see the difference as shown in the attached image.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_I was expecting the slicer to use the settings of the 1st object for both objects as it seemed like this would not print correctly with 2 different layer heights at the same level. Is this actually a feature and intended this way or is it a bug.  I can work around this problem, but thought I would bring it up in case it is a bug.
* _Actual Results_I get 2 different layer heights for the objects as can be seen in the attached screenshot
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Slic3rPE_LayerQuestion-1.41.0-beta.zip](https://github.com/prusa3d/Slic3r/files/2314955/Slic3rPE_LayerQuestion-1.41.0-beta.zip)

bubnikv, 2018-09-01:
Frankly I don't see a problem here. Slic3r is perfectly able to print each object with its layer height settings. Why do you think it is a problem?

geodave810, 2018-09-01:
You can close this issue then.  I thought it might be a problem.