prusa3d/PrusaSlicer #1161: Prusa MK2.5 profile is missing silent_mode=1

petrzelka, 2018-08-24:
### Version
1.41.0-beta

### Operating system type + version
OS independent

### Behavior
Latest slic3r now generates M73 codes to report progress, but for MK2.5 it doesn't generate silent codes (Q and S), only normal (P and R)
When my MK2.5 is running in silent mode and printing such gcode (from sd card) it reports "68:15R" all the time. 
Later I discovered it's because silent support checkbox is not checked in printer default (system) profile.
It would be great if you can set this by default to avoid WTFs of other guys.

Thanks!

bubnikv, 2018-09-01:
I have to check with the firmware guys.

bubnikv, 2018-09-27:
The firmware 3.4.0 shall handle the M73 Pxx Rxx correctly.

> Later I discovered it's because silent support checkbox is not checked in printer default (system) profile.
> It would be great if you can set this by default to avoid WTFs of other guys.

You should not need to check this checkbox for the MK2.5 and the 3.4.0 firmware. 

Closing.

closed 2018-09-27 by bubnikv