prusa3d/PrusaSlicer #1165: Supports on the build plate should have a solid bottom interface for better adhesion

magkopian, 2018-08-28:
### Version
1.41.0-beta

### Operating system type + version
Debian Stretch

### Behavior
When Slic3r generates supports on the build plate it doesn't use a solid bottom interface. This results to poor adhesion to the build plate, which often causes the supports to detach during a print.

[image]

As a workaround I often use a brim when I have to print something with a lot of supports. But this is not ideal because the brim touches only the support interface instead of the support itself. Additionally, it puts a brim around the object which is often unnecessary.

[image]

In my opinion, the bottom support interface for any supports that are in contact with the build plate should always be solid like a brim, or at least have an option to enable that.

mtmello2003, 2018-08-29:
+1
I agree.

supermerill, 2018-08-30:
duplicate of https://github.com/prusa3d/Slic3r/issues/1156
(6 rows below)

Area5142, 2018-09-03:
The need for support with _solid first layer_ depends on the used filament.
Printing mostly in PETG and flex _solid first layer_ will give me problems removing the support from the build plate as these materials sticks very well to the PEI surface.

A better solution would be to specify a first layer support density in percent, where 100% is solid.
It could be a new parameter in the filament tab, as this is filament depended.

magkopian, 2018-09-04:
@Area5142 When you print pretty much any object the first layer is always solid anyway, how printing supports in any different that this? If you have difficulty removing supports from the build plate when you print in PETG then you should have the same problem with removing the object itself.

Massa0815, 2018-09-04:
+1 That would help very much for my cheep Asian PEI sheet ;)

Area5142, 2018-09-04:
@magkopian a one layer high print is difficult to remove as it flexes with the steel sheet and only can be removed with a scraper. Taller models will have something to _grip on_, be more stiff and easier to remove just bending the steel sheet (MK2.5/MK3).
If I try to pull or bend the PEI sheet to release the support it will just break and leave the the solid first layer behind.

A filament parameter, specifying the layer density for the first one/two support layers will solve the issue for a range of filaments.

There could be other solutions, like a minimum support contact area/first layer area. When support footprint gets small they are easily overthrown, even in PETG.

magkopian, 2018-09-04:
> a one layer high print is difficult to remove as it flexes with the steel sheet and only can be removed with a scraper.

@Area5142 Isn't that the case for a non-solid interface layer as well? From my personally experience, I think that if a support breaks and only thing that remains behind is the interface layer in both cases you need a scraper. By the way, if you use a sheath around your supports it minimizes the chances of that happening.

Area5142, 2018-09-04:
@magkopian my experience with PETG is that the non-solid support releases the build plate with the model. If the bond gets stronger the support will break before releasing the PEI surface and leave the solid layer behind.

rongith, 2019-09-20:
I would have liked a 100% solid bottom for a case I encounter.

I printed a part made of ABS. It was very difficult to remove the  remaining support from the textured PEI sheet. With a smooth sticked PEI sheet, I never had an issue because as I could remove small residues of support with a scrapper. I can't do it on a textured sheet as the scrapper can't go well under the support. I had to solve all the remaining support with acetone to remove it.

If I had a 100% solid bottom, I could remove it as a whole thin sheet of ABS.

[image]

domtisdell, 2019-10-22:
+1 for % first layer support density

ANTALIFE, 2019-10-30:
+1 
Yup have the exact same issue right now. Tried playing around with modifiers but look like what you can do with them has been cut down compared to Slic3r

JamesLear92, 2020-06-24:
God, please yes. I just can't print with the current supports at all.

neophyl, 2020-06-24:
Give SuperSlicer a go, its built on Prusa Slicer so will be familiar to you and it can do that.  Under Support Material > Solid First Layer.

JamesLear92, 2020-06-27:
Awesome, thank you so much

anymouse404, 2020-08-09:
I would also like this option. I print PETG on smooth sheet and the zigzag pattern is very hard to remove. If it was a solid layer or ideally zigzagged thick trace, it would be possible to peel it without having to scrape off tiny stuck particles.

Whitehawk2000, 2020-12-06:
The problem still exists and makes the PE Slic3r unusable for serious printing with ABS.
If i remember it right the original Slic3r years ago did not have this stupid problem.
If someone activates the brim he is doing it for a purpose and not for any silly support-generation to override it.

bubnikv, 2021-02-24:
Implemented with fcb714cd241f0847522b9aac57da6b62acdd0152

New parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv