prusa3d/PrusaSlicer #1166: Slicing produces some inverted layers (solid and empty are flipped)

pentacular, 2018-08-29:
### Version
1.41.0-alpha2+linux64

### Operating system type + version
Debian

### Behavior
Some layers are inverted -- holes become solid and solids become holes.

Load model and slice for mk2 at 0.20 mm normal, mk3 at 0.20.
(mk2 at 0.35 works as expected, mk3 at 0.15 works as expected).

Slic3r claims that the model is manifold and without error.

I have a number of similar issues with missing layers that I can provide if useful.

3d object view
[image]

sliced view
[image]

bad layer -- the second layer -- immediately above 'good layer' and almost an inverse of it.
[image]

good layer -- the first layer
[image]

#### STL/Config (.ZIP) where problem occurs
[short_curve.zip](https://github.com/prusa3d/Slic3r/files/2332390/short_curve.zip)
[config.zip](https://github.com/prusa3d/Slic3r/files/2332397/config.zip)

bubnikv, 2018-08-30:
@FidelCapo will it help if you process the STL through a netfabb fixing service?

pentacular, 2018-08-31:
Sorry, I misdiagnosed it -- admesh doesn't help.

I've reduced it down to a small stl which reproduces the original problem.
[short_curve_debug.zip](https://github.com/prusa3d/Slic3r/files/2340428/short_curve_debug.zip)

Passing it through https://service.netfabb.com/ does produce an stl which doesn't reproduce the problem, but it also provides no feedback on what it fixed or what error might be present.

makeprintable reports no issues in the stl.

Are there any stl analysis programs or services that should be able to identify what's causing the problem?

If there is a problem with the STL then I'd like to fix the code that produced it.
If not, I'd like slic3r to operate correctly on it.

Passing everything through netfabb to hope that magic happens isn't a very satisfying solution. :)

pentacular, 2018-08-31:
Cutting the bottom 0.1 or top 0.1 off that short_curve_debug model fixes the problem in the pieces produced.

It feels like this might be an issue in slic3r which is avoided by any slight adjustment.

Let me know if you have any suggestions for diagnosis.

pentacular, 2018-09-01:
Ah, I think you're right -- scaling by 1.00001 seems to fix the problem :)

bubnikv, 2018-09-27:
We hoped that Slic3r-1.41.1-beta will fixed this issue, but it did not.

bubnikv, 2018-10-05:
It will be fixed with 1.41.1-rc. Closing.

closed 2018-10-05 by bubnikv