prusa3d/PrusaSlicer #1167: Bug: huge unnecessary internal supports when slicing with sparse infill and single perimeter

DeepSOIC, 2018-08-29:
### Version
1.41.0+beta+win64

### Operating system type + version
Windows 10 64-bit

### Behavior

my goal: print this object FAAAST! That is, thick nozzle, and use as little material as possible.

Problem: when slicing with single perimeter and sparse infill, massive dense infill bulk is added around all sloping walls, regardless of slope direction

[image]

For comparison, same settings but two perimeters:
 
[image]

Actually, I wouldn't mind a setting, that would the threshold angle for such inner support to appear.

#### STL/Config (.ZIP) where problem occurs

[single perimeter test.zip](https://github.com/prusa3d/Slic3r/files/2333059/single.perimeter.test.zip)

supermerill, 2018-08-30:
It's the "anchor" for the fill.
I added a parameter to let user change it in my fork https://github.com/supermerill/Slic3r
(Anchor solid infill by Xmm , in the fill tab)
Please write your feedback if it solve your problem or not.

DeepSOIC, 2018-09-25:
@supermerill 
Sorry for lack of feedback. I printed the thing and moved on to other duties.

I have now tried your fork, but it doesn't fix the problem. Also, the infill appears to be much denser than the setting, and not at 45 degree angle. Essentially the whole part is filled with this dense infill, except for a little bit near the top where I get islands of correct extremely-sparse infill. I don't know what's going on there.
[image]
I've downloaded Slic3r++_1.41.1_win32_20180910.zip, changed my settings that have worked with two perimeters to just 1, and got this. I tried setting both "anchor solid infill" values to zero, but got no change in slice result whatsoever.

supermerill, 2018-09-26:
@DeepSOIC

**please use the issue tracker in my fork, not this one, if you're reporting an issue about my fork.**
I don't want the prusa team to be bothered by my problems/mistakes.

untick "suporting dense layer"  
set "Anchor solid infill by Xmm" to 0

[image]

bubnikv, 2018-09-27:
Duplicate of https://github.com/prusa3d/Slic3r/issues/668
Closing this one, it will still be referenced from https://github.com/prusa3d/Slic3r/issues/668

closed 2018-09-27 by bubnikv