prusa3d/PrusaSlicer #1169: API not being sent to Octoprint

jwynn6, 2018-08-29:
### Version
Slic3rPE 1.40.1-Beta
Octoprint 1.3.9

### Operating system type + version
Arch Linux

### Behavior
When attempting to connect to octoprint from the printer settings tab, an error is encountered.  Octoprint is returning a 403 error that reads  "No API Key provided".  I have indeed included the API key in the interface so it's not that there isn't one being provided. 

When running the same version on my Mac, this issue does not occur. It seems to be something specifically with the Linux builds.

bubnikv, 2018-08-30:
Our tester @FidelCapo found out, that the pasting of an API key does not work correctly on Linux when pasted using a right mouse button. If you copy using Ctrl-V, it works. Would you please confirm that?

jwynn6, 2018-08-30:
Well paint me orange and call me an extruder. That absolutely worked. I've no idea why the behavior would be different but it sure is. Using Ctrl-V instead of the right mouse button let it work like a champ.

bubnikv, 2018-09-27:
Fixed in Slic3r-1.41.1-beta

closed 2018-09-27 by bubnikv