prusa3d/PrusaSlicer #1171: Gizmos for Y-axis and X-axis

tneff03, 2018-08-30:
### Version
1.41.0+win64

### Operating system type + version
Windows 7 Professional 64-bit

### Behavior

_Is this a new feature request?_ ->Yes

1.41.0 is an excellent update to Slic3r PE! I really appreciate the new "place on face" feature and the addition of the rotate about z gizmo!

In addition to the future addition of the side panel from PrusaControl, I would like to request rotation gizmos for x and y. Similar to other slicers such as Ultimaker Cura, Markforged Eiger, Raise3D ideaMaker, Windows 3D Builder, etc.

I'm imagining a setup where the rotate button is clicked, bringing up red, blue, and green rings around the respective axes. 
[image]

The user clicks on whichever ring they'd like to manipulate and that ring transforms into your very useful rotation gizmo now oriented in the correct plane allowing the user to freehand it or to snap to every 45 or 5 degrees depending on mouse position.
[image]

**Bonus points** for allowing the user to toggle between rotation relative to the bed coordinates or the local (model) coordinates.

bubnikv, 2018-09-27:
It is being worked on by @enricoturri1966

bubnikv, 2018-12-12:
The rotation gizmos in X and Y will be part of Slic3r 1.42 alpha, which is to be released soon.
Closing.

closed 2018-12-12 by bubnikv