prusa3d/PrusaSlicer #1175: No red, green,yellow Button MacOS

Medalknight, 2018-08-31:
### Version
1.41.0+

### Operating system type + version
MacOS 10.13.6

### Behavior
* _Describe the problem_
No red, green or yellow buttons to operate the window of slic3r.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
no

ctgreybeard, 2018-09-01:
Related ... I used Option-Green Dot (to maximize the window) and the top bar gets hidden under the title bar. Which makes it impossible to move because you can't grab the top of the window ...
[image]

ctgreybeard, 2018-09-01:
I found a solution ...

1. Close Slic3r
1. In ~/Library/Application Support/Slic3rPE edit the file slic3r.ini
1. Change two lines
    main_frame_pos = 120,70
    main_frame_size = 1188,890
1. Use values that work for you, these worked for me.

The problem seems to be that the `main_frame_pos` `Y` value gets set to negative (mine was -47) which sets it above the bottom of the menu bar.

FidelCapo, 2018-09-03:
We know about this issue and @vojtechkral working on a repair.

vojtechkral, 2018-09-04:
Yep, probably a duplicate of #1116

commit efd1b624 (2018-09-05): Fix window size persistence
Fixes #1116
Fixes #1175
commit 0ac26217 (2018-09-13): Fix window size persistence
Fixes #1116
Fixes #1175
commit 7258c597 (2018-09-14): Fix window size persistence
Fixes #1116
Fixes #1175
closed 2018-09-14 by bubnikv